raspberrypi/pico-sdk #1711: Need help for use I2C Slave mode

pierrotm777, 2024-05-22:
Hello,

I try to create a Hitec telemetry simulator and i need to define my I2C1 port as slave on the address 0x08.
I use this code:
`#include "hardware/i2c.h"
#include "hardware/irq.h"
#include "hardware/gpio.h"

uint8_t address = 0x08;

#define I2C_INTR_MASK_RD_REQ 0x00000020

void setup() {
  Serial.begin(9600);
  while (!Serial)
    ;
  i2c_init(i2c1, 100000L);
  i2c_set_slave_mode(i2c1, true, address);
  gpio_set_function(2, GPIO_FUNC_I2C);
  gpio_set_function(3, GPIO_FUNC_I2C);
  gpio_pull_up(2);
  gpio_pull_up(3);
  i2c1->hw->intr_mask = I2C_INTR_MASK_RD_REQ;
  irq_set_exclusive_handler(I2C1_IRQ, handler);
  irq_set_enabled(I2C1_IRQ, true);
}

void loop() {
}

void handler() {
  i2c1->hw->clr_rd_req;
  Serial.print("\nRequest from address: 0x");
  Serial.println(address);
  uint8_t packet[7] = {0x18, 0x17, 0x0, 0x0, 0x0, 0x0, 0x18};

  //i2c_write_raw_blocking(i2c1, packet, 7);
  
  for (uint8_t i = 0; i < 7; i++) {
    i2c_write_byte_raw(i2c1, packet[i]);
    Serial.print(packet[i], HEX);
    Serial.print(" ");
  }
}`
But if i use my logique analyser on the pins 2 et 3 i receive nothing.
Can you help me to find my error please ?

Thanks,

Pierre

lurch, 2024-05-22:
Your usage of `setup()` ad `loop()` makes it look like you're using one of the Arduino environments? That's not something that we support ourselves. Can you update your example to just use "pure" `pico-sdk` code please?
Also have you looked at https://github.com/raspberrypi/pico-examples/tree/master/i2c/slave_mem_i2c ?

pierrotm777, 2024-05-22:
[hitec.zip](https://github.com/raspberrypi/pico-sdk/files/15401444/hitec.zip)
This is my code for my tests