raspberrypi/pico-sdk #1716: Allow redirecting malloc functions to user defined functions

JNTeledyne, 2024-05-28:
I came across a case where we wanted to add custom memory debug/tracking code for malloc.  Since pico sdk already wraps malloc we need a way of calling our code from the wrapped pico version of malloc. I had 2 ideas to accomplish this.

 1. Add CMAKE switch to turn off building of pico_malloc so user could implement wrap functions on their own.  
 2. Create new PICO_REDIRECT_MALLOC define, if enabled it would replace REAL_FUNC, with REDIRECT_MALLOC_FUNC.  

My preference is option 2 as it allows user to still get the functionally in the pico_malloc wrapped functions and just extend it with more functionality.  

I have my suggested changes at https://github.com/JNTeledyne/pico-sdk/commit/399cdae0f21a1bb58e709156eeb74a2b002062c6  

Another good use of this is if someone wanted to have fun and implement a slab allocator or write their own version of malloc.

kilograham, 2024-07-21:
Note that if you don't include `pico_malloc` in your link libraries, then it won't get overridden, but note that `pico_malloc` is included by default by `pico_stdlib`

The simplest thing to do is to set `SKIP_PICO_MALLOC=1` at the top of your CMakeLists.txt, then it won't get included, you can 

Alternatively you can create your own `pico_malloc` library priort to pico_sdk_init()

kilograham, 2024-07-21:
closing this, as i dont think the extra define is necessary

closed 2024-07-21 by kilograham