raspberrypi/pico-sdk #1718: cyw43_arch_init stuck in infinite loop

tomosnelli, 2024-05-29:
Currently working on project that uses pico_cyw43_arch_lwip_sys_freertos library.
Upon calling cyw43_arch_init within a FreeRTOS task, cyw43_arch_init ever exists and gets stuck in an infinite loop on line 59 ~ 60 at function multicore_fifo_pop_blocking_inline() in file pico-sdk/src/rp2_common/pico_multicore/multicore.c.

Apologies if this is an already known issue or is a config error.

I'll post my main program along with the config files for further inspection.

pico_w_freertos_ble.c
```C
#include "pico/stdlib.h"
#include "pico/cyw43_arch.h"
#include "pico/btstack_cyw43.h"
#include "FreeRTOS.h"
#include "task.h"

void vTaskBlink( void * pvParameters )
{
    if( cyw43_arch_init() )
    {
        printf("Failed to run cyw43_arch_init");
    }

    for( ;; )
    {
        printf("blink loop");
        cyw43_arch_gpio_put( CYW43_WL_GPIO_LED_PIN, 1 );
        vTaskDelay( pdMS_TO_TICKS( 250 ) );
        cyw43_arch_gpio_put( CYW43_WL_GPIO_LED_PIN, 0 );
        vTaskDelay( pdMS_TO_TICKS( 250 ) );
    }

    cyw43_arch_deinit();
}

void vLaunch( void )
{
    BaseType_t xStatus;

    xStatus = xTaskCreate( vTaskBlink, "BLINK", 1024, NULL, configMAX_PRIORITIES - 4, NULL );

    if( xStatus != pdPASS )
    {
        printf("Error: xTaskCreate failed");
        exit 1;
    }
    else
    {
        vTaskStartScheduler();
    }
}

int main()
{
    stdio_init_all();
    printf("init stuff");

    vLaunch();

    for( ;; );

    return 0;
}
```

CMakeList.txt
```cmake
# Changing from pico to pico w board so running some blinky stuff

cmake_minimum_required(VERSION 3.13)

[... 342 lines cut ...]
#define TCP_CWND_DEBUG              LWIP_DBG_OFF
#define TCP_WND_DEBUG               LWIP_DBG_OFF
#define TCP_FR_DEBUG                LWIP_DBG_OFF
#define TCP_QLEN_DEBUG              LWIP_DBG_OFF
#define TCP_RST_DEBUG               LWIP_DBG_OFF
#define UDP_DEBUG                   LWIP_DBG_OFF
#define TCPIP_DEBUG                 LWIP_DBG_OFF
#define PPP_DEBUG                   LWIP_DBG_OFF
#define SLIP_DEBUG                  LWIP_DBG_OFF
#define DHCP_DEBUG                  LWIP_DBG_OFF

#if !NO_SYS
  #define TCPIP_THREAD_STACKSIZE      1024
  #define TCPIP_THREAD_PRIO           3
  #define DEFAULT_THREAD_STACKSIZE    1024
  #define DEFAULT_RAW_RECVMBOX_SIZE   8
  #define TCPIP_MBOX_SIZE             8
  #define LWIP_TIMEVAL_PRIVATE        0

  // not necessary, can be done either way
  #define LWIP_TCPIP_CORE_LOCKING_INPUT 1
#endif

#endif /* __LWIPOPTS_H__ */
```

HarterHorst, 2024-06-11:
What happens if you put the the following lines into main just after the stdio_init_all() and remove the cyw43_arch_deinit call from the vTaskBlink function? 

In my code I only init the cyw43 driver once in the main function and I never touch it again.
```
int main()
{
    stdio_init_all();
    if( cyw43_arch_init() )
        {
            printf("Failed to run cyw43_arch_init");
        }
    ...
```

peterharperuk, 2024-07-24:
I tested this and it seemed to work ok. I assume we've fixed the problem elsewhere.

closed 2024-07-24 by peterharperuk