raspberrypi/pico-sdk #1719: How to write to flash safely

ArvinHou, 2024-05-30:
I‘m using freertos(SMP) + lvgl .
There is 3 tasks.

```
    TaskHandle_t lvgl_task_handle;
    xTaskCreate(lv_timer_task_handler, "lvgl_task", 4096, NULL, (tskIDLE_PRIORITY + 4), &lvgl_task_handle);
    vTaskCoreAffinitySet(lvgl_task_handle, (1 << 0));

     TaskHandle_t video_flush_handler;
     xTaskCreate(video_flush_task, "video_flush", 4096, NULL, (tskIDLE_PRIORITY + 3), &video_flush_handler);
     vTaskCoreAffinitySet(video_flush_handler, (1 << 1));

    TaskHandle_t main_task_handler;
    xTaskCreate(main_task, "main_task", 4096, NULL, (tskIDLE_PRIORITY + 2), &main_task_handler);
    vTaskCoreAffinitySet(main_task_handler, (1 << 0));
```
The program can run successfully.
When a button click on screen. change a global value.Then write flash in main_task.

```
    uint32_t ints = save_and_disable_interrupts();
    flash_range_erase(FLASH_TARGET_OFFSET, FLASH_SECTOR_SIZE);
    flash_range_program(FLASH_TARGET_OFFSET, tmp, FLASH_PAGE_SIZE);
    restore_interrupts (ints);

```
 when code running flash_range_erase , the program crash ,But I can't see any logs。
Use Clion debug pico swd, the pc point running to 0xfffffffe .

I also tried the UF2 program method, Still failing.

I also tried using `multicore_lockout_start_blocking` `multicore_lockout_end_blocking`.But failed.Maybe there are some errors in my code.

How to write flash safely?
Can someone help me?
Thanks !

warcow105, 2024-05-30:
Have a look at the sdk documentation under pico_flash.  There are mechanisms for doing this, I have never used them though.  There is also the option of using the method you already have, but you need to make sure to use the victim init function on the secondary core.  I am not sure how that will work with freertos, but the pico_flash functions mention freertos and what to look out for.

kilograham, 2024-05-30:
best to visit https://forums.raspberrypi.com/viewforum.php?f=143 for questions

closed 2024-05-30 by kilograham

ArvinHou, 2024-05-30:
> Have a look at the sdk documentation under pico_flash. There are mechanisms for doing this, I have never used them though. There is also the option of using the method you already have, but you need [...]

Thank for reply.I searched carefully again and found that there are some functions  more suitable in pico_flash.
Put `flash_safe_execute_core_init();` at the begin of code in core 1.
When I want to write flash,using `flash_safe_execute` like bellow:
`flash_safe_execute(write_flash,&background_arc_color,2000);`
Well, it worked！