raspberrypi/pico-sdk #1725: No MQTT transmission after some time

HarterHorst, 2024-06-09:
| Pico SDK Version : 1.5.1
| lwIP Release     : 2.2.0d
---------------------------------
System Info:
```
Version: 7.95.49 (2271bb6 CY) CRC: b7a28ef3 Date: Mon 2021-11-29 22:50:27 PST Ucode Ver: 1043.2162 FWID 01-c51d9400
cyw43 loaded ok, mac d8:3a:dd:2e:36:38
API: 12.2
Data: RaspberryPi.PicoW
Compiler: 1.29.4
ClmImport: 1.47.1
Customization: v5 22/06/24
```
----

Hi all,

I am working on a larger project with the Pico and I'd like to use MQTT. I experienced some problems with the LwIP MQTT function using the PICO W in no_os mode. I have written a simple script that pushes a MQTT message every minute as a kind of alive packet. This works as intended, but after some time the transmission stops. I've turned on the debug functions to get an idea, but I can't find the cause. All parameters seem to be fine and the statistics show no memory shortage or other resource related problems.

I have observed the following behaviour.

1. The message "tcp_output: nothing to send (00000000)" appears every second, so I have 60 of these messages in the log between the various MQTT messages. After some time the number of messages reduces to just one message between the MQTT messages. Compare line 1816 and following with line 2014.
2. The mqtt callback function returns the following error: "mqtt_pub_request_cb: Publish result: -3 which indicates a timeout problem due to the documentation. see line 4253.
3. The IP stack seems fine as I can ping the device.

Any ideas what is wrong here or any ideas how to get more information from the debug system? code is below. minicom log is attached.

regards
Andi

```
[minicom.cap.zip](https://github.com/user-attachments/files/15753690/minicom.cap.zip)
#define LWIP_PLATFORM_ASSERT(message) custom_assert_handler(message)

#define MQTT_BROKER_IP_ADDR "192.168.0.3"
#define MQTT_TOPIC_ALIVE "home/IoT/rp2040/alive"
#define MQTT_BROKER_PORT 1883
#define MQTT_CLIENT_ID "RP2040-mmWave"

#define TimeBetweenAlivePackets 1000*60*1           // Every minute

#include <stdio.h>
#include <string.h>
#include <time.h>

// Pico Standard libraries
#include "pico/stdlib.h"
#include "pico/cyw43_arch.h"

// lwip libraries
#include "lwip/apps/mqtt.h"
#include "lwip/timeouts.h"
#include "lwip/init.h"
#include "lwip/stats.h"

// my own functions
#include "networkinfo.h"
#include "showversions.h"

const char *WIFI_SSID = "<REMOVED>";
const char *WIFI_PASSWORD = "<REMOVED>";

ip_addr_t mqtt_ip;

struct mqtt_connect_client_info_t mqtt_client_info;
char AliveMessage[] = "Alive";

err_t err;
uint8_t DeviceStatus;

[... 247 lines cut ...]
    memset(&mqtt_client_info, 0, sizeof(mqtt_client_info));
    mqtt_client_info.client_id = MQTT_CLIENT_ID;
    mqtt_client_info.keep_alive = 0; // 15 is default

    mqtt_client_info.will_topic = NULL;
    mqtt_client_info.will_msg = NULL;
    mqtt_client_info.will_qos = 0;
    mqtt_client_info.will_retain = 0;

    IP4_ADDR(&mqtt_ip, 192, 168, 0, 3);

    mqtt_client_t *mqtt_client = mqtt_client_new();
    mqtt_connect(mqtt_client);
    
    if (!add_repeating_timer_ms (TimeBetweenAlivePackets, timer_cb_SendAlivePacket, mqtt_client, &timer_system_alive)) {
        printf ("Error creating timer.\n");
    }

    while (true) {
        cyw43_arch_poll();
        sys_check_timeouts();
    }
    return 0;
}
```

jaguilar, 2024-07-24:
FWIW, I also had some issue with MQTT seemingly hanging up after about 23 hours of running. I ended up deciding to simply use the hardware watchdog. What I do is publish with a requirement for the server to acknowledge. If triple the expected publish period passes without the expected response coming back, the device reboots. Depending on your data accuracy needs, possibly this could work for you?

Another possibility: what if you restart the MQTT client each hour? What a pain right? But then you would not miss any data collections, if that fixes your problem?

Someday, when I have several minutes, I might like to implement the FreeRTOS IP stack on the Pico wifi driver, since maintenance in FreeRTOS seems a little more active.

pingufreak, 2024-08-13:
https://github.com/raspberrypi/pico-sdk/issues/1106

Increase MEMP_NUM_SYS_TIMEOUT

peterharperuk, 2024-11-12:
> Increase MEMP_NUM_SYS_TIMEOUT

What he said. Closing this as it's old and it's probably just a bug in your code. We should add the MQTT example to pico-examples. Please reopen if it's still a problem.

closed 2024-11-12 by peterharperuk