raspberrypi/pico-sdk #1741: error using  pico_setup.sh on Debian 12.5

j75, 2024-06-22:
Hello,

I was trying to follow in instructions from "Chapter 1. Quick Pico Setup" from getting-started-with-pico.pdf - running **pico_setup.sh** ended with
```
[  0%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  0%] No download step for 'ELF2UF2Build'
[  0%] Linking ASM executable bs2_default.elf
arm-none-eabi-gcc: fatal error: cannot read spec file 'nosys.specs': No such file or directory
compilation terminated.
make[2]: *** [pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/build.make:97: pico-sdk/src/rp2_common/boot_stage2/bs2_default.elf] Error 1
make[2]: Target 'pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/build' not remade because of errors.
```
And then
```
make[1]: Target 'blink/all' not remade because of errors.
make: *** [Makefile:91: all] Error 2
```

What should be done?

kilograham, 2024-06-22:
Best to ask/search on the forums for troubleshooting: https://forums.raspberrypi.com/viewforum.php?f=143

closed 2024-06-22 by kilograham

lurch, 2024-06-24:
Also, Chapter 1 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf hopefully makes it clear that the "Quick Pico Setup" script is only to be used if you're running on a Raspberry Pi.