raspberrypi/pico-sdk #1753: pico_add_extra_outputs( ) puts map file in wrong directory for add_subdirectory( ) projects

smithps, 2024-07-03:
I am using Visual Studio with VisualGDB to make a complex project and my Cmakelists.txt was getting a bit out of hand. I then created a sub folder to split my project into pieces. I found that the .map file for my executable that was in the subfolder was found in the directory above where the other extra output files were.

I have recreated the issue by simply using the "Hello Serial" example, then adding another copy of "Hello Serial" called "Hello2 Serial".  

The contents of my top level Cmakelists.txt is as so.

```
cmake_minimum_required(VERSION 3.12)
include(${PICO_SDK_PATH}/pico_sdk_init.cmake)

project(Hello C CXX ASM)
pico_sdk_init()

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
add_executable(hello_serial
        hello_serial.c
        )

# pull in common dependencies
target_link_libraries(hello_serial pico_stdlib)

# create map/bin/hex/uf2 file etc.
pico_add_extra_outputs(hello_serial)

# add url via pico_set_program_url
# example_auto_set_url(hello_serial)

add_subdirectory(Hello2)
```

The subdirectory Hello2 contains an almost identical Cmakelists.txt with hello_serial replaced with Hello2_serial everywhere,... The outputs I get are as follows:

```
ProjectDir
   |
   |__ VisualGDB
              |
              |___ Debug
                         |
                         |___ hello_serial.bin
                         |___ hello_serial.dis
                         |___ hello_serial.elf
                         |___ hello_serial.elf.map
                         |___ hello_serial.hex
                         |___ hello_serial.uf2
                         |___ hello2_serial.elf.map         <--  Expected to be in folder "Hello2"
                         |___ Hello2
                                    |___ hello2_serial.bin
                                    |___ hello2_serial.dis
                                    |___ hello2_serial.elf
                                    |___ hello2_serial.hex
                                    |___ hello2_serial.uf2
                
     
```

peterharperuk, 2024-07-03:
Does it do that when you build from the command line? It doesn't on Linux at least.

smithps, 2024-07-03:
There is the added complication that the build tools are "Sysprogs" specific versions of CMake and Ninja, running on a Windows 10 Machine, however when I build manually using the same tools in a Command prompt the map file still ended up in the wrong place. I wouldn't have thought that the Build tools would affect things, but maybe the difference is between Linux and Windows. 

I can post any debug/output logs if required, but I thought the example I gave was straightforward enough - can you try building on a Windows platform?

peterharperuk, 2024-07-04:
When I build on Windows using ninja I don't seem to get map files. Odd.

lurch, 2024-07-04:
The map files seem to be added at https://github.com/raspberrypi/pico-sdk/blob/develop/src/CMakeLists.txt#L59 ?

lurch, 2024-07-04:
Ahhh, looks like this might be related to https://github.com/raspberrypi/pico-sdk/pull/432#issuecomment-846449239 ?

smithps, 2024-07-04:
I am not by any stretch of the imagination familiar with Cmakelists, Cmake or Ninja but I did come to the same conclusion as @lurch  that it's dealt with seperately from the other additional output files  - and I also think it's related to Line 59 in the CMakelists.txt file 

I just don't have the knowledge to be able to figure out what might be a potential fix, so for now I will live with the file being in the "wrong" place.  Thanks for looking.

lurch, 2025-07-23:
Fix has been merged into `develop`

PR #2584 (2025-07-23): Fix location of map file on Ninja
Switch to using CMake generator expressions to ensure the `.elf.map` file is in the same location as the `.elf` file.

Fixes #1753
closed 2025-07-23 by lurch