raspberrypi/pico-sdk #1762: Add ability to provide custom memory allocation for LWIP

Memotech-Bill, 2024-07-08:
In order to resolve [an issue](https://github.com/Memotech-Bill/PicoBB/issues/26#issuecomment-2209320163) that has been raised against the Pico version of BBC BASIC, I would like to be able to provide my own memory allocation for LWIP, replacing the routines in either `${PICO_LWIP_PATH}/src/core/mem.c` and / or  `${PICO_LWIP_PATH}/src/core/memp.c`, however I cannot find any way to do this.

While there are CPP macro definitions such as `MEMP_MEM_MALLOC` and `MEM_LIBC_MALLOC` which select different memory allocators, there does not seem to be anything that allows providing user routines.

Using `#define` macros in `lwipopts.h` (or another include file) to rename the functions does not work since it renames both the function calls and the function definitions.

I have tried using the linker `--wrap=...` option, but that does not work as the symbols are defined in object files that are explicitly included in the link command, rather than being undefined symbols to be found by a library search.

Since the compilation of the lwip code is controlled by `pico-sdk/src/rp2_common/pico_lwip/CMakeLists.txt` it should be possible to add this feature to the SDK without requiring any modification to the lwip sub-module.

One possible approach would be to define three extra import libraries:

* **pico_lwip_nomem** - Omits mem.c and memp.c
* **pico_lwip_mem** - Provides mem.c
* **pico_lwip_memp** - Provides memp.c

Another option would be to define CMake variables `PICO_LWIP_NO_MEM` and `PICO_LWIP_NO_MEMP` to control the inclusion / exclusion of `mem.c` and `memp.c` in the **pico_lwip** library.

kilograham, 2026-09-30:
Fixed by #2405: lwIP 2.2.1 adds `MEM_CUSTOM_ALLOCATOR`. Define it and `MEM_CUSTOM_MALLOC`/`FREE`/`CALLOC` in `lwipopts.h`, and add `MEMP_MEM_MALLOC=1` to route the pools through it too. With `pico_cyw43_arch_lwip_threadsafe_background` it is called from IRQ context, so it must be IRQ-safe.

Memotech-Bill, 2026-09-30:
Thanks.

However, it turns out that what I need is the ability to customise `memp_malloc_pool()` and `memp_free_pool()`. Making these use `mem_malloc()` and `mem_free()` does not provide what I need.

My [work around](https://github.com/Memotech-Bill/PicoBB/tree/master/src/lwip) was to clone the `CMakeLists.txt` file from `pico-sdk/lib/src/core`, fix up the source paths to refer to the original pico-sdk files for all except `memp.c`, for which I supplied my own custom version.