raspberrypi/pico-sdk #1764: char vs unsigned char

Gavin-Perry, 2024-07-15:
I got stuck finding what my problem was with Serial.peek()
Documentation says it returns -1 if no char available
if (Serial.peek() >0) {   // works fine
BUT
char MyChar;
MyChar = Serial.peek();
if (MyChar > 0) was driving me crazy since MyChar == 255
I've got unsigned char if I wanted that type what happened to the normal type for char?
I was doing it this way because if the char is lower case I pass it to a different routine than if it's uppercase.
I get how I COULD do it but still had to ask about char being signed (sometimes, depending on compiler?)

kilograham, 2024-07-15:
Forums here are a good place for programming questions https://forums.raspberrypi.com/viewforum.php?f=143

closed 2024-07-15 by kilograham

lurch, 2024-07-15:
Also, it looks like you're using Arduino, which isn't something that Raspberry Pi officially support ourselves.