raspberrypi/pico-sdk #1770: Add PICO_PANIC_NO_STRINGS_ON_TARGET option

korneliuszo, 2024-07-23:
It allows in my code to remove 7kB of rodata.
It's more critical on copy_to_ram configuration

peterharperuk, 2024-07-23:
7kB?! Really?

kilograham, 2024-07-23:
Firstly, presumably, this only affects debug builds (release builds are probably already smaller)

Secondly, i think setting `PICO_PANIC_FUNCTION=` (i.e. empty) will do this already

korneliuszo, 2024-07-23:
@peterharperuk 
7kB on basic www with picow-http
@kilograham 
I have
`set(CMAKE_BUILD_TYPE "RelWithDebInfo")`
configured

korneliuszo, 2024-07-23:
PICO_PANIC_FUNCTION ... I suspect doesn't help as it is different compilation unit (header function prototype) so it needs provide string to call even nop function

kilograham, 2024-07-23:
> PICO_PANIC_FUNCTION ... I suspect doesn't help as it is different compilation unit (header function prototype) so it needs provide string to call even nop function

I don't know what you mean here

Add `PICO_PANIC_FUNCTION=` to your target_compile_definitions in CMake

korneliuszo, 2024-07-23:
I've made benchmarks
```
[no panic flags]$ arm-none-eabi-nm picopocket-sw.elf | grep __end__
20027e30 N __end__
[PICO_PANIC_FUNCTION=]$ arm-none-eabi-nm picopocket-sw.elf | grep __end__
20027da0 N __end__
[PICO_PANIC_NO_STRINGS_ON_TARGET]$ arm-none-eabi-nm picopocket-sw.elf | grep __end__
20025f60 N __end__

```