raspberrypi/pico-sdk #1782: arm-none-eabi-g++ showing error in WSL

Ganesharyal24894, 2024-08-06:
When I follow the documentation to build pico-examples in WSL it shows below error for some reason. This happens when I run 'make' command after executing 'cmake' command. The C++ compiler is not supposed to be check while building the firmware right? So why is this happening?

```
CMake Error at /usr/share/cmake-3.22/Modules/CMakeTestCXXCompiler.cmake:62 (message):
  The C++ compiler

    "/usr/bin/arm-none-eabi-g++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/user/pico/pico-examples/build/pioasm/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_e8d3a/fast && gmake[3]: Entering directory '/home/user/pico/pico-examples/build/pioasm/CMakeFiles/CMakeTmp'
    /usr/bin/gmake  -f CMakeFiles/cmTC_e8d3a.dir/build.make CMakeFiles/cmTC_e8d3a.dir/build
    gmake[4]: Entering directory '/home/user/pico/pico-examples/build/pioasm/CMakeFiles/CMakeTmp'
    Building CXX object CMakeFiles/cmTC_e8d3a.dir/testCXXCompiler.cxx.o
    /usr/bin/arm-none-eabi-g++    -o CMakeFiles/cmTC_e8d3a.dir/testCXXCompiler.cxx.o -c /home/user/pico/pico-examples/build/pioasm/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    Linking CXX executable cmTC_e8d3a
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_e8d3a.dir/link.txt --verbose=1
    /usr/bin/arm-none-eabi-g++ CMakeFiles/cmTC_e8d3a.dir/testCXXCompiler.cxx.o -o cmTC_e8d3a
    /usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): in function `exit':
    /build/newlib-pB30de/newlib-3.3.0/build/arm-none-eabi/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/exit.c:64: undefined reference to `_exit'
    collect2: error: ld returned 1 exit status
    gmake[4]: *** [CMakeFiles/cmTC_e8d3a.dir/build.make:99: cmTC_e8d3a] Error 1
    gmake[4]: Leaving directory '/home/user/pico/pico-examples/build/pioasm/CMakeFiles/CMakeTmp'
    gmake[3]: *** [Makefile:127: cmTC_e8d3a/fast] Error 2
    gmake[3]: Leaving directory '/home/user/pico/pico-examples/build/pioasm/CMakeFiles/CMakeTmp'

```

will-v-pi, 2024-08-09:
It looks like your build is trying to use the `arm-none-eabi-g++` compiler to build pioasm, when it should be using the native `g++` compiler, which is very odd. Could you try this again with the new 2.0.0 sdk and see if you get the same issue?

The documentation has also had some major overhauls, so could you confirm what documentation you're following?

Ganesharyal24894, 2024-08-10:
I threw my windows laptop that day itself...

lurch, 2024-08-21:
@Ganesharyal24894 In that close is it safe to close this issue?

Ganesharyal24894, 2024-08-21:
Yeah go ahead

closed 2024-08-22 by lurch