raspberrypi/pico-sdk #1788: Building with the pico-sdk requires gcc v9.1 or later

mattytrentini, 2024-08-09:
Building MicroPython with GCC 8.5 raises a build error:

```bash
In file included from micropython/ports/rp2/machine_pwm.c:32,
                 from micropython/extmod/machine_pwm.c:50:
micropython/lib/pico-sdk/src/rp2_common/hardware_pwm/include/hardware/pwm.h: In function 'pwm_set_irq_mask_enabled':
micropython/lib/pico-sdk/src/rp2_common/hardware_pwm/include/hardware/pwm.h:658:55: error: expected ',' before ')' token
     static_assert(PWM_IRQ_WRAP_1 == PWM_IRQ_WRAP_0 + 1);
```

This is because GCC versions ["prior to v9.1 do not support the single-argument `static_assert`"](https://www.gnu.org/software/gnulib/manual/html_node/assert_002eh.html).

I've confirmed that building with GCC v12.2 resolves the issue but I haven't tried with earlier versions (given the documentation I would _assume_ it works with v9.1 but I have yet to try). 

I suggest that either the code should be changed to add the second argument to `static_assert` - and thus work with earlier gcc versions - or document that v9.1 or later is required.

Grazfather, 2024-08-14:
I just hit this as well with up-to-date Xcode.

I worked around it simply by adding an empty message.

```diff
diff --git a/src/rp2_common/hardware_pwm/include/hardware/pwm.h b/src/rp2_common/hardware_pwm/include/hardware/pwm.h
index d394c72..d4da7e5 100644
--- a/src/rp2_common/hardware_pwm/include/hardware/pwm.h
+++ b/src/rp2_common/hardware_pwm/include/hardware/pwm.h
@@ -655,7 +655,7 @@ static inline void pwm_set_irq_mask_enabled(uint32_t slice_mask, bool enabled) {
         hw_clear_bits(&pwm_hw->inte, slice_mask);
     }
 #else
-    static_assert(PWM_IRQ_WRAP_1 == PWM_IRQ_WRAP_0 + 1);
+    static_assert(PWM_IRQ_WRAP_1 == PWM_IRQ_WRAP_0 + 1, "");
     uint irq_index = PWM_DEFAULT_IRQ_NUM() - PWM_IRQ_WRAP_0;
     if (enabled) {
         hw_set_bits(&pwm_hw->irq_ctrl[irq_index].inte, slice_mask);
```

kilograham, 2024-08-19:
fixed in develop

closed 2024-08-20 by kilograham