raspberrypi/pico-sdk #1789: PICO_BOARD_HEADER_DIRS not working with SDK v2.0

Memotech-Bill, 2024-08-09:
Attempting to build [PicoBB](https://github.com/Memotech-Bill/PicoBB).

My [CMakeLists.txt](https://github.com/Memotech-Bill/PicoBB/blob/d0723ecbf290803ffceaab94b9f0920e2f0c0a8d/src/pico/CMakeLists.txt#L30) file includes:

````
set(PICO_BOARD_HEADER_DIRS ${CMAKE_CURRENT_LIST_DIR}/../../boards)
````

And the [boards folder](https://github.com/Memotech-Bill/PicoBB/tree/master/boards) contains:

````
pi@raspberrypi:~/pico/PicoBB/console/pico_w $ ls ../../boards/
pico+w.h  vgaboard_cut.h  vgaboard_cut_w.h  vgaboard_sd.h  vgaboard_sd_w.h  vgaboard_serial.h  vgaboard_serial_w.h
````

This worked with previous versions of the SDK, however after downloading SDK v2.0 I now obtain:

````
cd build && PICO_SDK_PATH=/home/pi/pico/pico-sdk \
	cmake -DPICO_BOARD=pico+w -DSTDIO=USB+UART -DLFS=Y -DFAT=N -DSOUND=SDL \
		-DSTACK_CHECK=4 -DMIN_STACK=Y -DPRINTER= -DSERIAL_DEV=1 \
		-DCYW43=BACKGROUND -DBBC_SRC=../../BBCSDL -DGRAPH= -DOS_RAM= -DUSB_CON=0 \
		-DNET_HEAP=1  --no-warn-unused-cli -S ../../../src/pico -B .
Not searching for unused variables given on the command line.
Using PICO_SDK_PATH from environment ('/home/pi/pico/pico-sdk')
PICO_SDK_PATH is /home/pi/pico/pico-sdk
Target board (PICO_BOARD) is 'pico+w'.
CMake Error at /home/pi/pico/pico-sdk/cmake/generic_board.cmake:37 (message):
  Unable to find definition of board 'pico+w' (specified by PICO_BOARD):

     Looked for pico+w.h in /home/pi/pico/pico-sdk/cmake/../src/boards/include/boards (additional paths specified by PICO_BOARD_HEADER_DIRS)
     Looked for pico+w.cmake in /home/pi/pico/pico-sdk/cmake/../src/boards (additional paths specified by PICO_BOARD_CMAKE_DIRS)
Call Stack (most recent call first):
  /home/pi/pico/pico-sdk/cmake/pico_pre_load_platform.cmake:70 (include)
  /home/pi/pico/pico-sdk/pico_sdk_init.cmake:87 (include)
  /home/pi/pico/pico-sdk/external/pico_sdk_import.cmake:84 (include)
  CMakeLists.txt:17 (include)

-- Configuring incomplete, errors occurred!
````

peterharperuk, 2024-08-09:
> -DPICO_BOARD=pico+w

Did you mean to write the following? -DPICO_BOARD=pico_w

Memotech-Bill, 2024-08-09:
> > -DPICO_BOARD=pico+w
> 
> Did you mean to write the following? -DPICO_BOARD=pico_w

No. I have a custom pico+w.h which includes definitions for both Pico and Pico_W.

Found my own solution.

PICO_BOARD_HEADER_DIRS now has to be set before:

include($ENV{PICO_SDK_PATH}/external/pico_sdk_import.cmake)

I was setting it afterwards.

peterharperuk, 2024-08-09:
Great - can we close this then?

Memotech-Bill, 2024-08-09:
Resolved

closed 2024-08-09 by Memotech-Bill