raspberrypi/pico-sdk #1790: panic.c not building on bullseye

Memotech-Bill, 2024-08-09:
````
[ 27%] Building C object CMakeFiles/bbcbasic.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_platform_panic/panic.c.obj
/home/pi/pico/pico-sdk/src/rp2_common/pico_platform_panic/panic.c: In function 'panic':
/home/pi/pico/pico-sdk/src/rp2_common/pico_platform_panic/panic.c:34:5: warning: implicit declaration of function 'pico_default_asm' [-Wimplicit-function-declaration]
     pico_default_asm (
     ^~~~~~~~~~~~~~~~
/home/pi/pico/pico-sdk/src/rp2_common/pico_platform_panic/panic.c:45:26: error: expected ')' before ':' token
             "push {lr}\n"
                          ^
                          )
/home/pi/pico/pico-sdk/src/rp2_common/pico_platform_panic/panic.c:53:9:
         :
         ~                 
make[3]: *** [CMakeFiles/bbcbasic.dir/build.make:399: CMakeFiles/bbcbasic.dir/home/pi/pico/pico-sdk/src/rp2_common/pico_platform_panic/panic.c.obj] Error 1
````

Memotech-Bill, 2024-08-09:
Found the problem.

The source file **panic.c** is missing:

````
#include "pico/platform/compiler.h"
````

As a result, the following definition is missing:

````
#define pico_default_asm(...) __asm (".syntax unified\n" __VA_ARGS__)
````

So the compiler thinks `pico_default_asm()` is a standard C function call, and the colons are a syntax error.

I just need work out how to fix that in my build until the SDK gets fixed.

Memotech-Bill, 2024-08-09:
Messy, but adding the following line to my **CMakeLists.txt** file works around the issue for now:

````
  target_compile_options(${PROJECT_NAME} PUBLIC -include pico/platform/compiler.h)
````

kilograham, 2024-08-09:
Ah, good spot, only happens with `PICO_PANIC_FUNCTION` defined which is how we missed it

kilograham, 2024-11-20:
this (header issue) was already fixed by an earlier commit, so closing. Note i did make some changes in passing (#2062)

closed 2024-11-20 by kilograham