raspberrypi/pico-sdk #1793: pico-examples pico_w httpd build fails on Windows -- makefsdata.py is generating invalid variable name

ndabas, 2024-08-10:
When trying to do a full build of pico-examples on Windows, it fails on [the httpd example](https://github.com/raspberrypi/pico-examples/tree/master/pico_w/wifi/httpd) with:

```
In file included from C:/Program Files/Raspberry Pi/Pico SDK v2.0.0/pico-sdk/lib/lwip/src/apps/http/fs.c:39:
P:/pico-examples/build/pico_w/wifi/httpd/generated/pico_fsdata.inc:199:36: error: stray '\' in program
  199 | static const unsigned char data_img\rpi_png[] = {
      |                                    ^
```

So there's a `\` in the variable name.

The Python script that's generating this file is [assuming UNIX-style path separators](https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2_common/pico_lwip/tools/makefsdata.py#L107).

The fix would be to simply use `os.sep` instead of `"/"`. Let me know if this sounds reasonable and I can open a PR for it.

peterharperuk, 2024-08-13:
The code tries to replace invalid characters with "_". I guess it needs to do this for backslash as well as forward slash?

        # make a variable name
        var_name = str(file.relative_to(input_dir))
        var_name = var_name.replace(".", "_")
        var_name = var_name.replace("/", "_")
        data_var = f"data_{var_name}"
        file_var = f"file_{var_name}"

Apologies, I thought I tested this on Windows.

lurch, 2024-08-14:
> So maybe we can make it a bit more foolproof by substituting something like `r"[^a-zA-Z_\d]+"` in the string with `_` to get valid characters?

...which is exactly equivalent to `r"\W"` ? https://docs.python.org/3/library/re.html#regular-expression-syntax :wink:

ndabas, 2024-08-14:
> ...which is exactly equivalent to `r"\W"` ? https://docs.python.org/3/library/re.html#regular-expression-syntax 😉

Yep, that would be a good idea, I personally stay away from character classes which morph based on the locale in use -- but I guess `r"\w+"` with the `re.ASCII` flag will do nicely in this case.

Walter1604, 2024-10-31:
Has  this problem been resolved 

E:/Raspy/pico/pico-examples-eclipse/pico_w/wifi/httpd/generated/pico_fsdata.inc:203:36: error: stray '\' in program
  203 | static const unsigned char data_img\rpi_png[] = {

peterharperuk, 2024-11-01:
Looks like it. Please check the develop branch.

Walter1604, 2024-11-01:
Thanks how do I pull the branch    with git?

lurch, 2024-11-01:
> Thanks how do I pull the branch with git?

To check out the `develop` branch of `pico-sdk` into a new directory:
```
git checkout -b develop https://github.com/raspberrypi/pico-sdk.git
```

To update an _existing_ git checkout of `pico-sdk` from `master` to `develop`:
```
git checkout develop
git pull
```

Walter1604, 2024-11-02:
**### new problems using the develop branch**
[ 62%] [32mBuilding CXX object pico_w/bt/pan_lwip_http_server/CMakeFiles/picow_bt_example_pan_lwip_http_server_background.dir/E_/Raspy/pico/pico-sdk/src/rp2_common/pico_cxx_options/new_delete.cpp.obj[0m
arm-none-eabi-g++.exe: error: E:/Raspy/pico/pico-sdk/src/rp2_common/pico_cxx_options/new_dele: No such file or directory
arm-none-eabi-g++.exe: fatal error: no input files
compilation terminated.

lurch, 2024-11-02:
Did you do a `make clean` on your project after switching the pico-sdk branch?

peterharperuk, 2024-11-06:
Closing this as it's supposed to be fixed.

closed 2024-11-06 by peterharperuk

Walter1604, 2024-11-09:
Closing this as it's supposed to be fixed.

###  it is not fixed  still  the same problem using the  **develop branch**

lurch, 2024-11-09:
@Walter1604 the error message you've posted [above](https://github.com/raspberrypi/pico-sdk/issues/1793#issuecomment-2453098333) is different to the error message initially [reported](https://github.com/raspberrypi/pico-sdk/issues/1793#issue-2459087398) in this issue; so it's probably best if you open a new separate issue, with a demonstration of how to reproduce your problem. Thanks.

Walter1604, 2024-11-09:
Ok, I will do that.

PR #1841 (2024-08-20): Ensure makefsdata.py generates valid variable names
This PR:
- Updates makefsdata.py to strip out all characters that would be invalid in a C variable name.
- Adds a minor fix when getting the MIME type from the map to throw the exception we intend to, instead of `KeyError`.
- Fixes a minor indentation inconsistency which would bug me forever if not fixed.

There are a couple more issues in the script which can potentially be fixed, please let me know if you think it's worthwhile to do so and I will add these changes too:
- We could end up with duplicate variable names in some cases, if two or [... cut]