raspberrypi/pico-sdk #1798: Trouble building Bazel+pico_lwip_contrib_freertos

jaguilar, 2024-08-11:
Besides #1796, I'm still having some issues building with bazel and LWIP. I'm writing down the results of what investigation I've done for either myself or someone else to follow up on later.

I think the issue stems from a dependency that is not completely expressed in the Bazel-accepted DAG-ish way. So, pico_lwip_contrib_freertos in `lwip.BUILD` depends on pico_lwip_core. Let's look at one compile-time dependency chain there:

* pico_lwip_core -> mem.c
* mem.c -> arch.h
* arch.h -> sys_arch.h

All well and good, right? We're building pico_lwip_contrib_freertos, and that will provide sys_arch.h. Sadly not. The `includes` directories from `cc_library` get added to `-isystem` _only_ for the dependents of the library that defines it. Not the library's dependencies. So when we compile mem.c, `contrib/ports/freertos/include` is nowhere to be found on the compiler command-line.

This kind of makes sense. If a library is depending on a thing, it can't be necessary for the thing depended upon to require the library's existence. But you can actually do that in C. So I think we'll have to figure out how to express it. 

Intuitively, it seems to me like maybe the solution is something along the lines of "treat sys_arch.h as an interface and put it on the hdrs line of pico_lwip_core, but the definition is only available when pico_lwip_contrib_freertos is linked. But I'm not an expert on adapting C libraries to bazel so I'm open to other ideas.

jaguilar, 2024-08-11:
(Indeed, adding the sys directory to the includes of the core library does allow us to compile. But it's ?probably? not the right thing to do?

armandomontanez, 2024-08-11:
Sounds like ye ol' circular dependency. I'll try to carve out time to take a peek at this on Monday. I think there's some nuance around lwip's `NO_SYS` mode that needs to properly be hooked up as well.

jaguilar, 2024-08-11:
FWIW, I cracked this in my working copy by creating a header-only library that both pico_lwip_core and pico_lwip_contrib_freertos could depend on. No idea if this is proper but it seems to work okay.

armandomontanez, 2024-08-11:
Yep! 👍 That tends to be how I solve these kinds of cycles and was going to be my first step.

kilograham, 2024-11-19:
closing as fixed though it isn't yet tested

PR #1797 (2024-08-22): bazel build: Fix compilation bugs for Pico-W support
Fixes various minor errors with the bazel build files in the pico_lwip and pico_cyw43_driver directories that were causing compilation errors.

Fixes #1796 
Fixes #1798
closed 2024-11-19 by kilograham