raspberrypi/pico-sdk #1801: Contradictory SVD contents regarding CHAN_ABORT

jannic, 2024-08-11:
With version 2.0.0, the `access` specification of `CHAN_ABORT` was changed to `write-only` in `RP2040.svd`:

```
            <register>
                <name>CHAN_ABORT</name>
                <addressOffset>0x00000444</addressOffset>
                <description>Abort an in-progress transfer sequence on one or more channels</description>
                <resetValue>0x00000000</resetValue>
                <fields>
                    <field>
                        <name>CHAN_ABORT</name>
                        <description>Each bit corresponds to a channel. Writing a 1 aborts whatever transfer sequence is in progress on that channel. The bit will remain high until any in-flight transfers have been flushed through the address and data FIFOs.

                            After writing, this register must be polled until it returns all-zero. Until this point, it is unsafe to restart the channel.</description>
                        <bitRange>[15:0]</bitRange>
                        <access>write-only</access>
                    </field>
                </fields>
            </register>
```

I guess this was done because of RP2040-E13 and one should poll `BUSY` instead of `CHAN_ABORT`.

However, the `description` still says "After writing, this register must be polled until it returns all-zero." This can't be done on a write-only register.

lurch, 2024-08-12:
ping @matiasilva

kilograham, 2024-11-18:
ping @matiasilva @Wren6991

Wren6991, 2024-11-21:
@matiasilva I checked the upstream sources and this field (one-field register) is correctly marked as `sc`. Is something going wrong in the new header export?

matiasilva, 2025-02-12:
2.1.1 should have this fixed. The access type is reverted to `read-write` for any write/set clear fields but at the same time I've also added ` <modifiedWriteValues>oneToClear</modifiedWriteValues>` to help tooling see writing 1 does have special effects.

kilograham, 2025-02-18:
fix merged into develop

PR #2270 (2025-02-18): fix up access type of CHAN_ABORT register in RP2040.svd and RP2350.svd
fixes #1801