raspberrypi/pico-sdk #1802: sdk2.0.0: PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE=0 incompatible with CDC

UKTailwind, 2024-08-11:
As per the title, My application was using this parameter in previous releases (1.5, 1.6) with no issues. On rebuilding with sdk2.0.0 the application didn't work (see https://forums.raspberrypi.com/viewtopic.php?p=2243782#p2243782) and this was tracked down to this setting. Adding the parameter to the build of hello-usb proved the problem

kilograham, 2024-08-20:
I added `PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE=0` to `hello_usb` and it still enumerates fine (at least on Ubuntu)... @will-v-pi can you give it a spin on windows

Are you using stdio_usb or custom CDC?

will-v-pi, 2024-08-21:
I can confirm this is broken on Windows - I upgraded the `bcdUSB` to 210 to support WinUSB drivers for the reset interface, but this means Windows doesn't recognise the device without that enabled (`PICO_STDIO_USB_RESET_INTERFACE_SUPPORT_MS_OS_20_DESCRIPTOR`). I'll push a PR to switch `bcdUSB` back to `200` if that isn't enabled

kilograham, 2024-08-21:
So what exactly changed in 2.0.0 that broke this?

will-v-pi, 2024-08-21:
It was changing `bcdUSB` from `0x0200` to `0x0210` - when `bcdUSB` is `0x0210` on Windows then it requires the MS_OS_20 descriptors, which I added to the reset interface for driverless picotool on Windows.

So if you're not using the reset interface MS_OS_20 descriptors it needs to remain at `0x0200` else Windows does not recognise the USB device.

lurch, 2024-09-17:
I think this was fixed on `develop` by #1857 , so can this be closed now @will-v-pi ?

PR #1857 (2024-08-24): Fix stdio_usb with reset interface disabled on Windows
Only use bcdUSB = 0x0210 when reset interface and MS_OS_20 descriptor are enabled, otherwise Windows doesn't recognise the USB device

I can confirm this fixes #1802 on Windows
closed 2024-09-18 by will-v-pi