raspberrypi/pico-sdk #1804: pshell doesn't compile on SDK 2.0.0 targetting Pico

matsobdev, 2024-08-12:
When trying to compile https://github.com/lurk101/pshell (1.2.30) using SDK 2.0.0 and ARM GCC 13.3.1 targetting Pico and some definitions are unavailable:
```
In file included from W:\pshell\cc\cc.c:251:
W:\pshell\cc\cc_defs.h:23:23: error: 'GPIO_FUNC_XIP' undeclared here (not in a function); did you mean 'GPIO_FUNC_SIO'?
   23 |     {"GPIO_FUNC_XIP", GPIO_FUNC_XIP},
      |                       ^~~~~~~~~~~~~
      |                       GPIO_FUNC_SIO
W:\pshell\cc\cc_defs.h:77:17: error: 'clk_rtc' undeclared here (not in a function); did you mean 'clk_adc'?
   77 |     {"clk_rtc", clk_rtc},
      |                 ^~~~~~~
      |                 clk_adc
W:\pshell\cc\cc_defs.h:85:26: error: 'PICO_DEFAULT_I2C_INSTANCE' undeclared here (not in a function)
   85 |     {"i2c_default", (int)PICO_DEFAULT_I2C_INSTANCE},
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~
W:\pshell\cc\cc_defs.h:92:26: error: 'PICO_DEFAULT_SPI_INSTANCE' undeclared here (not in a function)
   92 |     {"spi_default", (int)PICO_DEFAULT_SPI_INSTANCE},
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~
W:\pshell\cc\cc_defs.h:109:21: error: 'TIMER_IRQ_0' undeclared here (not in a function); did you mean 'TIMER1_IRQ_0'?
  109 |     {"TIMER_IRQ_0", TIMER_IRQ_0},
      |                     ^~~~~~~~~~~
      |                     TIMER1_IRQ_0
W:\pshell\cc\cc_defs.h:110:21: error: 'TIMER_IRQ_1' undeclared here (not in a function); did you mean 'TIMER1_IRQ_1'?
  110 |     {"TIMER_IRQ_1", TIMER_IRQ_1},
      |                     ^~~~~~~~~~~
      |                     TIMER1_IRQ_1
W:\pshell\cc\cc_defs.h:111:21: error: 'TIMER_IRQ_2' undeclared here (not in a function); did you mean 'TIMER1_IRQ_2'?
  111 |     {"TIMER_IRQ_2", TIMER_IRQ_2},
      |                     ^~~~~~~~~~~
      |                     TIMER1_IRQ_2
W:\pshell\cc\cc_defs.h:112:21: error: 'TIMER_IRQ_3' undeclared here (not in a function); did you mean 'TIMER1_IRQ_3'?
  112 |     {"TIMER_IRQ_3", TIMER_IRQ_3},
      |                     ^~~~~~~~~~~
      |                     TIMER1_IRQ_3
W:\pshell\cc\cc_defs.h:115:17: error: 'XIP_IRQ' undeclared here (not in a function); did you mean 'OTP_IRQ'?
  115 |     {"XIP_IRQ", XIP_IRQ},
      |                 ^~~~~~~
      |                 OTP_IRQ
W:\pshell\cc\cc_defs.h:124:23: error: 'SIO_IRQ_PROC0' undeclared here (not in a function)
  124 |     {"SIO_IRQ_PROC0", SIO_IRQ_PROC0},
      |                       ^~~~~~~~~~~~~
W:\pshell\cc\cc_defs.h:125:23: error: 'SIO_IRQ_PROC1' undeclared here (not in a function)
  125 |     {"SIO_IRQ_PROC1", SIO_IRQ_PROC1},
      |                       ^~~~~~~~~~~~~
W:\pshell\cc\cc_defs.h:134:17: error: 'RTC_IRQ' undeclared here (not in a function); did you mean 'OTP_IRQ'?
  134 |     {"RTC_IRQ", RTC_IRQ},
      |                 ^~~~~~~
      |                 OTP_IRQ
In file included from W:\pshell\cc\cc.c:425:
W:\pshell\cc\cc_extrns.h:33:37: error: 'clocks_init' undeclared here (not in a function); did you mean 'clocks_hw_t'?
   33 |     {"clocks_init", 0, clk_defines, clocks_init, 0, 0, 0},
      |                                     ^~~~~~~~~~~
      |                                     clocks_hw_t
W:\pshell\cc\cc_extrns.h:100:38: error: 'i2c_hw_index' undeclared here (not in a function); did you mean 'i2c_get_index'?
  100 |     {"i2c_hw_index", 1, i2c_defines, i2c_hw_index, 0, 0, 0},
      |                                      ^~~~~~~~~~~~
      |                                      i2c_get_index
```
Manually hardcoding values allowes it to compile. But the switching to Pico2 and ARM and:
```
[  4%] Linking CXX executable pshell_usb.elf
E:/Download/Pico_2/arm-gnu-toolchain-13.3.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe:
Binary info must be in first 256 bytes of the binary
E:/Download/Pico_2/arm-gnu-toolchain-13.3.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe:
CMakeFiles/pshell_usb.dir/cc/cc.c.obj: in function `gen':
W:\pshell\cc/cc.c:2877:(.text.gen+0x5e4): undefined reference to `__wrap___aeabi_fcmple'
E:/Download/Pico_2/arm-gnu-toolchain-13.3.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe:
W:\pshell\cc/cc.c:2877:(.text.gen+0x5e8): undefined reference to `__wrap___aeabi_fcmpge'
E:/Download/Pico_2/arm-gnu-toolchain-13.3.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe:
W:\pshell\cc/cc.c:2892:(.text.gen+0x1088): undefined reference to `__wrap___aeabi_fsub'
E:/Download/Pico_2/arm-gnu-toolchain-13.3.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe:
[... 37 lines cut ...]
CMakeFiles/pshell_usb.dir/cc/cc.c.obj:(.rodata.fops+0x8): undefined reference to `__wrap___aeabi_i2f'
E:/Download/Pico_2/arm-gnu-toolchain-13.3.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe:
CMakeFiles/pshell_usb.dir/cc/cc.c.obj:(.rodata.fops+0xc): undefined reference to `__wrap___aeabi_f2iz'
E:/Download/Pico_2/arm-gnu-toolchain-13.3.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe:
CMakeFiles/pshell_usb.dir/cc/cc.c.obj:(.rodata.fops+0x10): undefined reference to `__wrap___aeabi_fadd'
E:/Download/Pico_2/arm-gnu-toolchain-13.3.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe:
CMakeFiles/pshell_usb.dir/cc/cc.c.obj:(.rodata.fops+0x14): undefined reference to `__wrap___aeabi_fsub'
E:/Download/Pico_2/arm-gnu-toolchain-13.3.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe:
CMakeFiles/pshell_usb.dir/cc/cc.c.obj:(.rodata.fops+0x18): undefined reference to `__wrap___aeabi_fmul'
E:/Download/Pico_2/arm-gnu-toolchain-13.3.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe:
CMakeFiles/pshell_usb.dir/cc/cc.c.obj:(.rodata.fops+0x1c): undefined reference to `__wrap___aeabi_fdiv'
E:/Download/Pico_2/arm-gnu-toolchain-13.3.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe:
CMakeFiles/pshell_usb.dir/cc/cc.c.obj:(.rodata.fops+0x20): undefined reference to `__wrap___aeabi_fcmple'
E:/Download/Pico_2/arm-gnu-toolchain-13.3.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe:
CMakeFiles/pshell_usb.dir/cc/cc.c.obj:(.rodata.fops+0x24): undefined reference to `__wrap___aeabi_fcmpgt'
E:/Download/Pico_2/arm-gnu-toolchain-13.3.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe:
CMakeFiles/pshell_usb.dir/cc/cc.c.obj:(.rodata.fops+0x28): undefined reference to `__wrap___aeabi_fcmplt'
E:/Download/Pico_2/arm-gnu-toolchain-13.3.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe:
CMakeFiles/pshell_usb.dir/cc/cc.c.obj:(.rodata.fops+0x2c): undefined reference to `__wrap___aeabi_fcmpge'
collect2.exe: error: ld returned 1 exit status
mingw32-make[2]: *** [CMakeFiles\pshell_usb.dir\build.make:1450: pshell_usb.elf] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:2287: CMakeFiles/pshell_usb.dir/all] Error 2
mingw32-make: *** [makefile:102: all] Error 2
```
Obviously building in separate dirs.

peterharperuk, 2024-08-12:
These look like issues with that project. You should probably raise an issue there?

matsobdev, 2024-08-12:
To be honest, it was motivated only to improve SDK if there is an issue. BTW, it compiles on SDK 1.5.1 just fine.

closed 2024-08-12 by matsobdev

lurch, 2024-08-12:
> To be honest, it was motivated only to improve SDK if there is an issue.

https://github.com/raspberrypi/pico-sdk/releases/tag/2.0.0 has a big list of things that have changed in the latest SDK release.

matsobdev, 2024-08-12:
Sounds fair, my bad.

kilograham, 2024-08-12:
Note those errors at the top should not happen - you started with clean directory for RP2040 too? it looks like you have a mix of old / new setup.

There is a change listed in the release notes: PICO_DEFAULT_XXX_INSTANCE is now PICO_DEFAULT_XXX_INSTANCE()

matsobdev, 2024-08-12:
Yes, there was a mixup - that first error list was from Pico2, Pico on SDK 2.0.0 outputs only:
```
In file included from ../cc/cc.c:251:
../cc/cc_defs.h:85:26: error: 'PICO_DEFAULT_I2C_INSTANCE' undeclared here (not in a function)
   85 |     {"i2c_default", (int)PICO_DEFAULT_I2C_INSTANCE},
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~
../cc/cc_defs.h:92:26: error: 'PICO_DEFAULT_SPI_INSTANCE' undeclared here (not in a function)
   92 |     {"spi_default", (int)PICO_DEFAULT_SPI_INSTANCE},
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../cc/cc.c:425:
../cc/cc_extrns.h:33:37: error: 'clocks_init' undeclared here (not in a function); did you mean 'clocks_hw_t'?
   33 |     {"clocks_init", 0, clk_defines, clocks_init, 0, 0, 0},
      |                                     ^~~~~~~~~~~
      |                                     clocks_hw_t
../cc/cc_extrns.h:100:38: error: 'i2c_hw_index' undeclared here (not in a function); did you mean 'i2c_get_index'? C object CMakeFiles/pshell_usb.dir/
  100 |     {"i2c_hw_index", 1, i2c_defines, i2c_hw_index, 0, 0, 0},
      |                                      ^~~~~~~~~~~~
      |                                      i2c_get_index
```
and `PICO_DEFAULT_XXX_INSTANCEs` are easily fixable with new definitions. Just compiling some examples on a RAM partition - starting with empty partition and to be sure, downloaded it again.

kilograham, 2024-08-12:
Ah gotcha - yes we have new macros you can use which work across both ro2040 and rp2350  whrrre there are different numbers of instances/irqs on the two chips - check release notes