raspberrypi/pico-sdk #1805: pico_atomic: atomic operations are not multicore-atomic

geurtv, 2024-08-12:
From SDK 2.0.0 release notes:
> Additional support for C11 atomic functions using spin lock number PICO_SPINLOCK_ID_ATOMIC.

and

> On RP2040, all functions are implemented via spinlock.
	
Well ... no? `PICO_SPINLOCK_ID_ATOMIC` doesn't seem to be used anywhere in the pico-sdk sources and spinlocks are nowhere to be found in atomic.c. 

The atomic operations in atomic.c call `atomic_lock()` which itself only calls `save_and_disable_interrupts()`, and if I'm not mistaken this only disables interrupts on the current cpu core.

kilograham, 2024-08-12:
Wow... what? how embarrassing... there was some merging of different implementations... clearly something went wrong in the merge!!!

kilograham, 2024-08-19:
#1805 merged into develop

closed 2024-08-19 by kilograham