raspberrypi/pico-sdk #1807: 'picotoolBuild' refuses to build

jimjag, 2024-08-12:
```
[  6%] Performing build step for 'picotoolBuild'
/Users/src/pico/pico-sdk/lib/mbedtls/library/bignum.c:1395:29: error: variable 't' set but not used [-Werror,-Wunused-but-set-variable]
    mbedtls_mpi_uint c = 0, t = 0;
                            ^
1 error generated.
make[5]: *** [lib/mbedtls/library/CMakeFiles/mbedcrypto.dir/bignum.c.o] Error 1
make[4]: *** [lib/mbedtls/library/CMakeFiles/mbedcrypto.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [pico-sdk/src/rp2040/boot_stage2/picotool/src/picotoolBuild-stamp/picotoolBuild-build] Error 2
make[1]: *** [pico-sdk/src/rp2040/boot_stage2/CMakeFiles/picotoolBuild.dir/all] Error 2
make: *** [all] Error 2
```

macOS

will-v-pi, 2024-08-13:
Is your mbedtls submodule up to date? This should be fixed in the version of mbedtls we’re now using in the SDK (https://github.com/Mbed-TLS/mbedtls/issues/7166).

You can update your submodules by running `git submodule update --init`

jimjag, 2024-08-13:
Forcing an update solved the problem.

lurch, 2024-08-30:
Sounds like this issue has been resolved, so closing.

closed 2024-08-30 by lurch