raspberrypi/pico-sdk #1813: LLVM: ld.lld: warning: cannot find entry symbol _start; not setting start address

matsobdev, 2024-08-13:
Both on RP2040 and RP2350, with Ninja or without this happens on Windows:
```
PICO_SDK_PATH is E:/Download/Pico_2/pico-sdk-2.0.0
Target board (PICO_BOARD) is 'pico2'.
Using board configuration from E:/Download/Pico_2/pico-sdk-2.0.0/src/boards/include/boards/pico2.h
Pico Platform (PICO_PLATFORM) is 'rp2350-arm-s'.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m33_clang'
Setting default C library to picolibc as LLVM appears to be using it
-- The C compiler identification is Clang 18.1.3
-- The CXX compiler identification is Clang 18.1.3
-- The ASM compiler identification is Clang
-- Found assembler: E:/Download/Pico_2/LLVM-ET-Arm-18.1.3-Windows-x86_64/bin/clang.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: E:/Download/Pico_2/LLVM-ET-Arm-18.1.3-Windows-x86_64/bin/clang.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: E:/Download/Pico_2/LLVM-ET-Arm-18.1.3-Windows-x86_64/bin/clang++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
-- Found Python3: C:/DEV/Python/python.exe (found version "3.8.10") found components: Interpreter
TinyUSB available at E:/Download/Pico_2/pico-sdk-2.0.0/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at E:/Download/Pico_2/pico-sdk-2.0.0/lib/btstack
cyw43-driver available at E:/Download/Pico_2/pico-sdk-2.0.0/lib/cyw43-driver
lwIP available at E:/Download/Pico_2/pico-sdk-2.0.0/lib/lwip
mbedtls available at E:/Download/Pico_2/pico-sdk-2.0.0/lib/mbedtls
-- Configuring done
-- Generating done
-- Build files have been written to: W:/pico2/pico2/build
[84/84] Linking CXX executable project_name.elf
ld.lld: warning: cannot find entry symbol _start; not setting start address
```
and this on Ubuntu 20.04:
```
mateush@mateush:~/pico2/pico2$ cmake -G Ninja -S ~/pico2/pico2 -B ~/pico2/pico2/build
PICO_SDK_PATH is /home/mateush/pico-sdk-2.0.0
Target board (PICO_BOARD) is 'pico2'.
Using board configuration from /home/mateush/pico-sdk-2.0.0/src/boards/include/boards/pico2.h
Pico Platform (PICO_PLATFORM) is 'rp2350-arm-s'.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m33_clang'
Setting default C library to picolibc as LLVM appears to be using it
-- The C compiler identification is Clang 18.1.3
-- The CXX compiler identification is Clang 18.1.3
-- The ASM compiler identification is Clang
-- Found assembler: /home/mateush/LLVM-ET-Arm-18.1.3-Linux-x86_64/bin/clang
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/mateush/LLVM-ET-Arm-18.1.3-Linux-x86_64/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/mateush/LLVM-ET-Arm-18.1.3-Linux-x86_64/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
-- Found Python3: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter 
TinyUSB available at /home/mateush/pico-sdk-2.0.0/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/mateush/pico-sdk-2.0.0/lib/btstack
cyw43-driver available at /home/mateush/pico-sdk-2.0.0/lib/cyw43-driver
lwIP available at /home/mateush/pico-sdk-2.0.0/lib/lwip
mbedtls available at /home/mateush/pico-sdk-2.0.0/lib/mbedtls
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mateush/pico2/pico2/build
mateush@mateush:~/pico2/pico2$ cmake --build ~/pico2/pico2/build
[2/84] Linking ASM executable pico-sdk/src/rp2350/boot_stage2/bs2_default.elf
ld.lld: warning: cannot find entry symbol _start; not setting start address
[84/84] Linking CXX executable project_name.elf
```
Similar, but quite not I guess.

matsobdev, 2024-08-13:
If I'm doing it right by adding it to `CMakeLists.txt`:
```
add_compile_options(
	-Wl,--entry=_stage2_boot
	)
```
or
```
cmake -G Ninja -S W:\pico2\pico2 -B W:\pico2\pico2\build -Wl,--entry=_stage2_boot
```
then it doesn't change a thing. 15.0.2 adds some extras (with or without above):
```
PICO_SDK_PATH is E:/Download/Pico_2/pico-sdk-2.0.0
Target board (PICO_BOARD) is 'pico2'.
Using board configuration from E:/Download/Pico_2/pico-sdk-2.0.0/src/boards/include/boards/pico2.h
Pico Platform (PICO_PLATFORM) is 'rp2350-arm-s'.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m33_clang'
Setting default C library to picolibc as LLVM appears to be using it
-- The C compiler identification is Clang 15.0.2
-- The CXX compiler identification is Clang 15.0.2
-- The ASM compiler identification is Clang
-- Found assembler: E:/Download/Pico_2/bin/clang.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: E:/Download/Pico_2/bin/clang.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: E:/Download/Pico_2/bin/clang++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
-- Found Python3: C:/DEV/Python/python.exe (found version "3.8.10") found components: Interpreter
TinyUSB available at E:/Download/Pico_2/pico-sdk-2.0.0/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at E:/Download/Pico_2/pico-sdk-2.0.0/lib/btstack
cyw43-driver available at E:/Download/Pico_2/pico-sdk-2.0.0/lib/cyw43-driver
lwIP available at E:/Download/Pico_2/pico-sdk-2.0.0/lib/lwip
mbedtls available at E:/Download/Pico_2/pico-sdk-2.0.0/lib/mbedtls
-- Configuring done
-- Generating done
-- Build files have been written to: W:/pico2/pico2/build
[84/84] Linking CXX executable project_name.elf
ld.lld: warning: cannot find entry symbol _start; not setting start address

ld.lld: warning: section type mismatch for .ram_vector_table
>>> CMakeFiles/project_name.dir/E_/Download/Pico_2/pico-sdk-2.0.0/src/rp2_common/pico_runtime_init/runtime_init.c.obj:(.ram_vector_table): SHT_PROGBIT
S
>>> output section .ram_vector_table: SHT_NOBITS

ld.lld: warning: section type mismatch for .heap
>>> CMakeFiles/project_name.dir/E_/Download/Pico_2/pico-sdk-2.0.0/src/rp2_common/pico_crt0/crt0.S.obj:(.heap): SHT_PROGBITS
>>> output section .heap: SHT_NOBITS

ld.lld: warning: section type mismatch for .stack
>>> CMakeFiles/project_name.dir/E_/Download/Pico_2/pico-sdk-2.0.0/src/rp2_common/pico_crt0/crt0.S.obj:(.stack): SHT_PROGBITS
>>> output section .stack_dummy: SHT_NOBITS
```

kilograham, 2024-11-21:
merged into develop

closed 2024-11-21 by kilograham