raspberrypi/pico-sdk #1817: Interrupt handler rename to CMSIS standard broken in 2.0.0

tannewt, 2024-08-13:
In CircuitPython we use a mix of standard CMSIS irq handler names (like `HardFault_Handler`) and pico sdk specific ones (like `isr_dma_0`).

The rename to standard names doesn't work in 2.0.0 because the IRQ table is now generated with an assembly macro (?) which doesn't appear to happen before the C preprocessor that renames the handlers.

Generator code:
https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2_common/pico_crt0/crt0.S#L55-L64

1.5.1 version:
https://github.com/raspberrypi/pico-sdk/blob/6a7db34ff63345a7badec79ebea3aaef1712f374/src/rp2_common/pico_standard_link/crt0.S#L43-L51

The RP2350 names aren't quite correct either because they end in IRQn:
https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2_common/cmsis/include/cmsis/rename_exceptions.h#L58

kilograham, 2024-08-14:
doh; our test for this uses PendSV_Handler which didn't get broken

kilograham, 2024-08-14:
@matiasilva can you check on the IRQ names in the SVD

kilograham, 2024-08-20:
merged into develop

PR #1821 (2024-08-20): Fix Cmsis irq regression
Fixes #1817
closed 2024-08-20 by kilograham