raspberrypi/pico-sdk #1834: Should we add pio_sm_set_pins_with_mask64 and pio_sm_set_pindirs_with_mask64?

peterharperuk, 2024-08-16:
If you try calling pio_sm_set_pins_with_mask or pio_sm_set_pindirs_with_mask with a gpio >= 32 you'll get a build error. You have to know to use pio_get_gpio_base when constructing the mask. Just wondering if it would be more obvious to add a 64bit version of these functions? We'd have to assert it was valid for the gpiobase?

kilograham, 2024-08-17:
ha, yes, and indeed the existing (non-mask) functions are just plain broken it seems

peterharperuk, 2024-08-19:
You mean pio_sm_set_pins and pio_sm_set_consecutive_pindirs? I hadn't noticed that. Might have to write some test code

StijnKuipers, 2024-09-24:
Hmm - I've just been hit with this one on a board I constructed. If any code needs to be tested I can run it here.

peterharperuk, 2024-09-24:
Which function are you trying to use?

StijnKuipers, 2024-09-25:
I was trying to get this bit of code to output, ported from a RP2040 product -> 

```
pio_sm_config c = toggleprog_program_get_default_config(offset);
pio_set_gpio_base(pio, 16);
    
pio_gpio_init(pio, LEDCLK);
pio_gpio_init(pio, LEDDAT);
pio_gpio_init(pio, LEDLAT);

#define ADJUST(x) ((x+16)%32) 
sm_config_set_out_pin_base(&c, ADJUST(LEDDAT));
sm_config_set_set_pin_base(&c, ADJUST(LEDDAT));
pio_sm_set_consecutive_pindirs(pio, sm, ADJUST(LEDDAT), 1, true);
pio_sm_set_consecutive_pindirs(pio, sm, ADJUST(LEDCLK), 2, true);

sm_config_set_sideset(&c, 2, false, false);

sm_config_set_sideset_pins(&c, ADJUST(LEDCLK));
sm_config_set_out_pins(&c, ADJUST(LEDDAT), 1);

sm_config_set_fifo_join(&c, PIO_FIFO_JOIN_TX);
sm_config_set_clkdiv(&c, clk_div);
sm_config_set_out_shift(&c, false, true, 32);

 pio_sm_init(pio, sm, offset, &c);
 pio_sm_set_enabled(pio, sm,true);
    ```

peterharperuk, 2024-11-19:
@StijnKuipers Just looked at your code again. It looks like your code is buggy.

The following functions handle pins >= 32, do not adjust your pin numbers when calling them.
sm_config_set_out_pin_base 
pio_sm_set_consecutive_pindirs
sm_config_set_sideset_pins
sm_config_set_out_pins

lurch, 2024-11-20:
> The following functions handle pins >= 32, do not adjust your pin numbers when calling them.
sm_config_set_out_pin_base

Doxygen at https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_pio/include/hardware/pio.h#L263 says `\param out_base 0-31 First pin to set as output` so I guess that might need updating to avoid confusion?

Also, would you get a compile-time error if you tried doing both `sm_config_set_out_pin_base(&c, 10)` and `sm_config_set_in_pin_base(&c, 40)` ?

kilograham, 2024-11-22:
merged into develop

PR #2074 (2024-11-20): Add some missing pio functions
Adds some 64 versions to work with rp2350b and more than 32 pins. Change cyw43 code to use the new function.

Fixes #1834
closed 2024-11-22 by kilograham