raspberrypi/pico-sdk #1836: MTB Support

TinyCircuits, 2024-08-16:
Hi,

I'm wondering if the new RP2350A2 supports Arm's **Micro Trace Buffer** (**MTB**): https://developer.arm.com/documentation/100231/latest/

The reason I ask is because the datasheet is a bit unclear. For example, pg. 125 of the datasheet says ["MTB: no MTB trace"](https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf#page=125) but [pg. 126](https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf#page=126) says

"
The **processor implements** the following optional interfaces:
• Arm TrustZone technology, using the Armv8-M Security Extension supporting Secure and Non-secure states
• Memory Protection Units (MPUs), which you can configure to protect regions of memory
• Floating-point arithmetic functionality with support for single precision arithmetic
• **Support for ETM and MTB trace**
"

The rest of the datasheet also has mentions of MTB indicating it is supported. Reading the processor ROM for MTB at address [0xE00FF01C](https://developer.arm.com/documentation/100230/0002/debug-and-trace-components/debug/debug-functionality/processor-rom-table-identification-and-entries) returns `0xFFF44002` indicating it isn't supported. The SDK also has no mentions of the keyword **MTB**.

Just curious as this would let me diagnose crashes for customers a little easier.

Thanks.

Wren6991, 2024-08-17:
No, we don't have MTB. The processor supports it but we don't implement it. The reason is that MTB requires either a dedicated SRAM instance, or muxing on the front of normal system SRAM which leads to unacceptable timing degradation.

As an alternative we have the trace FIFO (3.5.7.2) which captures TPIU output at up to 32 bits/cycle and streams to internal memory via the DMA. It functions as a standard trace capture device, similar to an external probe.

The reason for the MTB reference on page 125 is probably because ETM/MTB support are guarded by the same feature flag when configuring the Cortex-M33. I'll remove the MTB reference as it's misleading, thanks.

TinyCircuits, 2024-08-19:
@Wren6991 Understood. I did look at 3.5.7 and TPIU which captures ETM (and ITM) to an SRAM location if DMA is setup to do so. However, according to [table 97] on pg. 91(https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf#page=91), the most likely way for me to get that to work would be by connecting the `TRACECTL` pin to GND to enable to-fifo trace saving. There are two issues here, first, I can't figure out which GPIO pin is `TRACECTL` (Trace Control?), and second, I'm all out of pins so tying one to GND likely isn't possible.

Maybe I am miss understanding something though. Is there an all software way to enable TPIU to FIFO without connecting GPIO pins a certain way?

Thanks.

Wren6991, 2024-08-21:
`TRACECTL` there is referring to an output from the TPIU, on-chip. You don't need any GPIOs when tracing to the internal FIFO. If you set up the TPIU to output trace packets (without configuring any GPIOs), and then clear the `TRACE_CAPTURE_FIFO_FLUSH` bit on the trace FIFO controls, then data should start accumulating in the FIFO.

philiprehkop, 2024-11-07:
Did anybody make the ITM==>TPIU==>DMA chain run and could provide an example? I tried to make it run, but I did not get valid ITM data.
I described some details in https://forums.raspberrypi.com/viewtopic.php?t=378970