raspberrypi/pico-sdk #1837: Flash sizes larger than 2MB  compile with an flash size error

benseller, 2024-08-16:
If a program exceeds the default 2MB flash limit the following compiler error occurs:

/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: blink.elf section `.rodata' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 2121368 bytes

In this case a 4MB flash size was needed.

The board configuration header file has the flash size defined as:
#define PICO_FLASH_SIZE_BYTES (16 * 1024 * 1024)

The generated ./build/pico_flash_region.ld in the project directory  however still states the default flash size:
FLASH(rx) : ORIGIN = 0x10000000, LENGTH = (2 * 1024 * 1024)

When I change this to:
FLASH(rx) : ORIGIN = 0x10000000, LENGTH = (16 * 1024 * 1024)

the project compiles with no errors and a correct .uf2 file is generated.

Looks like the generation of ./build/pico_flash_region.ld in the project directory is incorrect.

OS: Xubuntu 24.04
pico-sdk: 2.0.0

Regards,

Ben Seller

kilograham, 2024-08-16:
You need to include
```
// pico_cmake_set_default PICO_FLASH_SIZE_BYTES = (16 * 1024 * 1024)
```

in your board header, like the other board headers

closed 2024-08-16 by kilograham

lurch, 2024-08-16:
The `check_board_header.py` in #1814 checks for `// pico_cmake_set_default PICO_FLASH_SIZE_BYTES = XXX` :slightly_smiling_face:

benseller, 2024-08-17:
Thanks. That did the job. 
Still strange that you need to change comment in a header file that already has a correct #define to compile without errors.

lurch, 2024-08-18:
I believe that the `#define` communicates the value to the "C world" and the `pico_cmake_set_default` communicates the value to the "CMake world". See https://github.com/raspberrypi/pico-sdk/blob/master/cmake/generic_board.cmake#L20-L30