raspberrypi/pico-sdk #1842: boot2_name missing in UF2 when compiled with sdk 2.0

codetiger, 2024-08-18:
I've a custom PCB based on RP2040 and tried upgrading the SDK to the latest version. What is observed is the device won't start when reset, but works perfectly when dropping UF2. This problem sounds familiar and I went through this when I first tried to run code on the custom PCB. I fixed the issue with the below lines 

```
#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
#define PICO_BOOT_STAGE2_CHOOSE_W25Q080 1
#define PICO_FLASH_SPI_CLKDIV 2

```

After compiling the code using SDK 2.0, the issue started again. Tried setting the same config in the CMakeLists.txt

```
add_compile_definitions(PICO_XOSC_STARTUP_DELAY_MULTIPLIER=64)
add_compile_definitions(PICO_BOOT_STAGE2_CHOOSE_W25Q080=1)
add_compile_definitions(PICO_FLASH_SPI_CLKDIV=2)

```

But does not solve the issue. 

Using picotool to see what's the difference in the uf2 below is what I found

```
codetiger@Mac-mini build % picotool info -a ~/Downloads/GameTiger.uf2
File /Users/codetiger/Downloads/GameTiger.uf2:

Program Information
 name:          GameTiger
 features:      USB stdin / stdout
 binary start:  0x10000000
 binary end:    0x10098d3c

Fixed Pin Information
 none

Build Information
 sdk version:       1.5.1
 pico_board:        gametiger_rp2040
 boot2_name:        boot2_w25q080
 build date:        Apr  3 2024
 build attributes:  Release
```

```
codetiger@Mac-mini build % picotool info -a GameTiger.uf2            
File GameTiger.uf2:

Program Information
 name:          GameTiger
 features:      USB stdin / stdout
 binary start:  0x10000000
 binary end:    0x10098d38

Fixed Pin Information
 none

Build Information
 sdk version:       2.0.0
 pico_board:        gametiger_rp2040
 build date:        Aug 18 2024
 build attributes:  Release

```
 
If you see closely the parameter `boot2_name ` is missing in the UF2 file.

codetiger, 2024-08-18:
Not sure why, but with SDK 2.0, it works only with 
`#define PICO_FLASH_SPI_CLKDIV 4`

kilograham, 2024-08-18:
> Not sure why, but with SDK 2.0, it works only with
#define PICO_FLASH_SPI_CLKDIV 4

that is certainly surprising.

codetiger, 2024-08-19:
After comparing my whole codebase, I found that in previous version I fixed this problem using 
`PICO_XOSC_STARTUP_DELAY_MULTIPLIER` set to 64. However that didn't fix the issue after upgrading to SDK 2.0. This time I also had to change the `PICO_FLASH_SPI_CLKDIV` to 4. There is no other change in my code.

codetiger, 2024-08-19:
Yes, am able to reproduce.

Overclocking Configuration
```
    vreg_set_voltage(VREG_VOLTAGE_1_20);
    sleep_ms(1);
    set_sys_clock_khz(266 * 1000, true);
```

Old SDK version worked with 
```
#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
#define PICO_FLASH_SPI_CLKDIV 2
```

New SDK version works only with 
```
#define PICO_FLASH_SPI_CLKDIV 4
```
and the `PICO_XOSC_STARTUP_DELAY_MULTIPLIER ` set to 64 does not have any effect in new sdk.

kilograham, 2024-08-25:
renamed the issue, since the missing binary info is now fixed

Wren6991, 2024-11-23:
This is running the RP2040 at double its rated frequency and the W25Q16JV at exactly its maximum frequency. I would guess your timing is marginal and the apparent fix of `PICO_XOSC_STARTUP_DELAY_MULTIPLIER` is pulling you just back from the edge by avoiding the warmup period where the XOSC has higher jitter.

The fact that the CLKDIV had to be increased from 2 to 4 for the new SDK is surprising but the startup code is different, so memory access during early boot will be different. Not conclusively unrelated to timing.

@kilograham I'm moving this to the next release as it's not yet clear that this is an SDK issue