raspberrypi/pico-sdk #1843: Windows CI builds failing on GitHub with "WARNING: Disassembly is not correct"

lurch, 2024-08-18:
See the CI build logs of any of the recent commits on the `develop` branch https://github.com/raspberrypi/pico-sdk/commits/develop/

```
[133/2153] Linking CXX executable test\pico_stdlib_test\pico_stdlib_test.elf
FAILED: test/pico_stdlib_test/pico_stdlib_test.elf 
cmd.exe /C "cd . && C:\ProgramData\chocolatey\bin\arm-none-eabi-g++.exe -mcpu=cortex-m0plus -mthumb -Og -g -Wl,--build-id=none -Wl,-Map=pico_stdlib_test.elf.map --specs=nosys.specs -Wl,--wrap=__clzsi2 -Wl,--wrap=__clzdi2 -Wl,--wrap=__ctzsi2 -Wl,--wrap=__popcountsi2 -Wl,--wrap=__popcountdi2 -Wl,--wrap=__clz -Wl,--wrap=__clzl -Wl,--wrap=__clzll -Wl,--wrap=__ctzdi2 -Wl,--wrap=__aeabi_idiv -Wl,--wrap=__aeabi_idivmod -Wl,--wrap=__aeabi_ldivmod -Wl,--wrap=__aeabi_uidiv -Wl,--wrap=__aeabi_uidivmod -Wl,--wrap=__aeabi_uldivmod -Wl,--wrap=__aeabi_dadd -Wl,--wrap=__aeabi_ddiv -Wl,--wrap=__aeabi_dmul -Wl,--wrap=__aeabi_drsub -Wl,--wrap=__aeabi_dsub -Wl,--wrap=__aeabi_cdcmpeq -Wl,--wrap=__aeabi_cdrcmple -Wl,--wrap=__aeabi_cdcmple -Wl,--wrap=__aeabi_dcmpeq -Wl,--wrap=__aeabi_dcmplt -Wl,--wrap=__aeabi_dcmple -Wl,--wrap=__aeabi_dcmpge -Wl,--wrap=__aeabi_dcmpgt -Wl,--wrap=__aeabi_dcmpun -Wl,--wrap=__aeabi_i2d -Wl,--wrap=__aeabi_l2d -Wl,--wrap=__aeabi_ui2d -Wl,--wrap=__aeabi_ul2d -Wl,--wrap=__aeabi_d2iz -Wl,--wrap=__aeabi_d2lz -Wl,--wrap=__aeabi_d2uiz -Wl,--wrap=__aeabi_d2ulz -Wl,--wrap=__aeabi_d2f -Wl,--wrap=sqrt -Wl,--wrap=cos -Wl,--wrap=sin -Wl,--wrap=tan -Wl,--wrap=atan2 -Wl,--wrap=exp -Wl,--wrap=log -Wl,--wrap=ldexp -Wl,--wrap=copysign -Wl,--wrap=trunc -Wl,--wrap=floor -Wl,--wrap=ceil -Wl,--wrap=round -Wl,--wrap=sincos -Wl,--wrap=asin -Wl,--wrap=acos -Wl,--wrap=atan -Wl,--wrap=sinh -Wl,--wrap=cosh -Wl,--wrap=tanh -Wl,--wrap=asinh -Wl,--wrap=acosh -Wl,--wrap=atanh -Wl,--wrap=exp2 -Wl,--wrap=log2 -Wl,--wrap=exp10 -Wl,--wrap=log10 -Wl,--wrap=pow -Wl,--wrap=powint -Wl,--wrap=hypot -Wl,--wrap=cbrt -Wl,--wrap=fmod -Wl,--wrap=drem -Wl,--wrap=remainder -Wl,--wrap=remquo -Wl,--wrap=expm1 -Wl,--wrap=log1p -Wl,--wrap=fma -Wl,--wrap=__aeabi_lmul -Wl,--wrap=__aeabi_fadd -Wl,--wrap=__aeabi_fdiv -Wl,--wrap=__aeabi_fmul -Wl,--wrap=__aeabi_frsub -Wl,--wrap=__aeabi_fsub -Wl,--wrap=__aeabi_cfcmpeq -Wl,--wrap=__aeabi_cfrcmple -Wl,--wrap=__aeabi_cfcmple -Wl,--wrap=__aeabi_fcmpeq -Wl,--wrap=__aeabi_fcmplt -Wl,--wrap=__aeabi_fcmple -Wl,--wrap=__aeabi_fcmpge -Wl,--wrap=__aeabi_fcmpgt -Wl,--wrap=__aeabi_fcmpun -Wl,--wrap=__aeabi_i2f -Wl,--wrap=__aeabi_l2f -Wl,--wrap=__aeabi_ui2f -Wl,--wrap=__aeabi_ul2f -Wl,--wrap=__aeabi_f2iz -Wl,--wrap=__aeabi_f2lz -Wl,--wrap=__aeabi_f2uiz -Wl,--wrap=__aeabi_f2ulz -Wl,--wrap=__aeabi_f2d -Wl,--wrap=sqrtf -Wl,--wrap=cosf -Wl,--wrap=sinf -Wl,--wrap=tanf -Wl,--wrap=atan2f -Wl,--wrap=expf -Wl,--wrap=logf -Wl,--wrap=ldexpf -Wl,--wrap=copysignf -Wl,--wrap=truncf -Wl,--wrap=floorf -Wl,--wrap=ceilf -Wl,--wrap=roundf -Wl,--wrap=sincosf -Wl,--wrap=asinf -Wl,--wrap=acosf -Wl,--wrap=atanf -Wl,--wrap=sinhf -Wl,--wrap=coshf -Wl,--wrap=tanhf -Wl,--wrap=asinhf -Wl,--wrap=acoshf -Wl,--wrap=atanhf -Wl,--wrap=exp2f -Wl,--wrap=log2f -Wl,--wrap=exp10f -Wl,--wrap=log10f -Wl,--wrap=powf -Wl,--wrap=powintf -Wl,--wrap=hypotf -Wl,--wrap=cbrtf -Wl,--wrap=fmodf -Wl,--wrap=dremf -Wl,--wrap=remainderf -Wl,--wrap=remquof -Wl,--wrap=expm1f -Wl,--wrap=log1pf -Wl,--wrap=fmaf -Wl,--wrap=malloc -Wl,--wrap=calloc -Wl,--wrap=realloc -Wl,--wrap=free -Wl,--wrap=memcpy -Wl,--wrap=memset -Wl,--wrap=__aeabi_memcpy -Wl,--wrap=__aeabi_memset -Wl,--wrap=__aeabi_memcpy4 -Wl,--wrap=__aeabi_memset4 -Wl,--wrap=__aeabi_memcpy8 -Wl,--wrap=__aeabi_memset8 -Wl,-LD:/a/pico-sdk/pico-sdk/build -Wl,--script=D:/a/pico-sdk/pico-sdk/src/rp2_common/pico_crt0/rp2040/memmap_default.ld -Wl,-z,max-page-size=4096 -Wl,--gc-sections -Wl,--wrap=sprintf -Wl,--wrap=snprintf -Wl,--wrap=vsnprintf -Wl,--wrap=printf -Wl,--wrap=vprintf -Wl,--wrap=puts -Wl,--wrap=putchar -Wl,--wrap=getchar @CMakeFiles\pico_stdlib_test.rsp -o test\pico_stdlib_test\pico_stdlib_test.elf  && cmd.exe /C "cd /D D:\a\pico-sdk\pico-sdk\build\test\pico_stdlib_test && C:\ProgramData\chocolatey\bin\arm-none-eabi-objdump.exe -h D:/a/pico-sdk/pico-sdk/build/test/pico_stdlib_test/pico_stdlib_test.elf > pico_stdlib_test.dis && C:\ProgramData\chocolatey\bin\arm-none-eabi-objdump.exe -d D:/a/pico-sdk/pico-sdk/build/test/pico_stdlib_test/pico_stdlib_test.elf >> pico_stdlib_test.dis && ( D:\a\pico-sdk\pico-sdk\build\_deps\picotool\picotool coprodis --quiet pico_stdlib_test.dis pico_stdlib_test.dis || "C:/Program Files/CMake/bin/cmake.exe" -E echo "WARNING: Disassembly is not correct" ) && cd /D D:\a\pico-sdk\pico-sdk\build\test\pico_stdlib_test && C:\ProgramData\chocolatey\bin\arm-none-eabi-objcopy.exe -Oihex D:/a/pico-sdk/pico-sdk/build/test/pico_stdlib_test/pico_stdlib_test.elf pico_stdlib_test.hex && cd /D D:\a\pico-sdk\pico-sdk\build\test\pico_stdlib_test && C:\ProgramData\chocolatey\bin\arm-none-eabi-objcopy.exe -Obinary D:/a/pico-sdk/pico-sdk/build/test/pico_stdlib_test/pico_stdlib_test.elf pico_stdlib_test.bin && cd /D D:\a\pico-sdk\pico-sdk\build\test\pico_stdlib_test && D:\a\pico-sdk\pico-sdk\build\_deps\picotool\picotool uf2 convert --quiet D:/a/pico-sdk/pico-sdk/build/test/pico_stdlib_test/pico_stdlib_test.elf pico_stdlib_test.uf2 --family rp2040 --abs-block""
WARNING: Disassembly is not correct
[134/2153] Building C object test/pico_stdio_test/CMakeFiles/pico_stdio_test_uart.dir/__/__/src/rp2_common/hardware_ticks/ticks.c.obj
[135/2153] Building C object test/pico_stdio_test/CMakeFiles/pico_stdio_test_uart.dir/__/__/src/rp2_common/hardware_xosc/xosc.c.obj
[136/2153] Building C object test/pico_stdio_test/CMakeFiles/pico_stdio_test_uart.dir/__/__/src/rp2_common/hardware_watchdog/watchdog.c.obj
[137/2153] Building ASM object test/pico_stdio_test/CMakeFiles/pico_stdio_test_uart.dir/__/__/src/rp2_common/hardware_divider/divider.S.obj
[138/2153] Building C object test/pico_stdio_test/CMakeFiles/pico_stdio_test_uart.dir/__/__/src/rp2_common/pico_runtime/runtime.c.obj
ninja: build stopped: subcommand failed.
Error: Process completed with exit code 1.
```

kilograham, 2024-08-18:
Not sure that that warning is entirely the cause, but @will-v-pi should take a look at this anyway... that said the "||" in the command looks suspiciously windows breaking.

Also, I would suggest that "WARNING: Disassembly is not correct" is not a very helpful message. It doesn't really describe what happened, or what to do about it". I might be tempted to leave it out (is this just that there were some garbage copro instructions, or that there was no picotool to call?)

lurch, 2024-08-19:
Fixed by #1846

PR #1846 (2024-08-19): Fix Windows CI build
Use powershell for the build, instead of bash - I can't test this works without this being merged, but running the CI commands locally in powershell works fine, so they should run correctly in GitHub actions. There seem to be some path escaping issues or something in bash on Windows, so using powershell instead as that's a more standard option on Windows generally

Also remove the unnecessary warning - it isn't necessary, as picotool no longer throws an error with garbage coprocessor instructions

Fixes #1843
closed 2024-08-19 by lurch