raspberrypi/pico-sdk #1848: Pico 2 board intermittently does not re-enumerate after programming with `picotool`.

konkers, 2024-08-19:
One of the 3 production Pico 2's we've been testing does not reliably re-enumerate after programming.  This was discovered while running through automated Pigweed testing.  The workflow I was using is the same as the "Standalone Raspberry Pi Pico" workflow from https://pigweed.dev/targets/rp2040/upstream.html with `--config=rp2040` replaced with `--config=rp2350`.  This workflow uses `picotool`.

Running these tests results in back-to-back program -> execute cycles.  I would occasionally see a failure with this one board where the board did not re-enumerate over USB after programming.  Observed on a USB analyzer, I saw the device disconnect as usual then not reconnect ([pico2-failure.tdc.zip](https://github.com/user-attachments/files/16665226/pico2-failure.tdc.zip) Beagle trace)

I did not get a chance to attach over SWD to see if the whole core was locked up or if it was executing code but stuck.  

None of our pre-productions A2 Pico 2s exhibited this issue.

The board has been returned to Chris at RPi for failure analysis.

lurch, 2024-09-09:
Could this be a similar problem to https://github.com/raspberrypi/pico-sdk/issues/1882 ? ping @will-v-pi

konkers, 2024-09-09:
Unfortunately I don't have an exact hash.  Looking back at my chat logs, I was seeing this on August 8th so The roll-up 2350 commit at `a861301071402fac84e5b980fb7189eb41a4c1c5` (https://pwrev.dev/228326) would be a good candidate.  

I was using an M1 Macbook Pro.  I believe the Pico 2 was behind 1 or 2 USB hubs.

liamfraser, 2024-09-10:
I ran out of space running the test suite on a Pi 4 with a 64GB SD card. So unless there's an easier way to reproduce this I'm not going to be able to. It seems fine with a pico-sdk application rebooted 10k times in a loop via picotool. So I'm going to close the issue.

closed 2024-09-10 by liamfraser