raspberrypi/pico-sdk #1849: TinyUSB submodule needs a bump in order to get Pico-PIO-USB working

lurch, 2024-08-20:
The version of the [TinyUSB](https://github.com/hathach/tinyusb) submodule included (in `lib/tinyusb`) in the `develop` branch of [pico-sdk](https://github.com/raspberrypi/pico-sdk) currently [pulls in](https://github.com/hathach/tinyusb/blob/master/tools/get_deps.py#L60) an older version of [Pico-PIO-USB](https://github.com/sekigon-gonnoc/Pico-PIO-USB), which is [incompatible](https://github.com/sekigon-gonnoc/Pico-PIO-USB/issues/128) with PIco SDK 2.0.0, and this currently means that none of the [USB Dual Mode examples](https://github.com/raspberrypi/pico-examples?tab=readme-ov-file#usb-dual-mode) are compatible with the current version of Pico-SDK.

The problem with Pico-PIO-USB has been [fixed](https://github.com/sekigon-gonnoc/Pico-PIO-USB/pull/126) and that newer version has also been [picked up](https://github.com/hathach/tinyusb/pull/2760) by TinyUSB, so if pico-sdk bumped to a newer version of TinyUSB, then I believe the USB Dual Mode examples _would_ then be compilable with the `develop` version of Pico-SDK.

lurch, 2024-08-30:
https://github.com/raspberrypi/pico-examples/issues/516 is a related TinyUSB issue.

liamfraser, 2024-11-19:
This will be fixed by moving to 0.17.0

kilograham, 2024-11-19:
merged into develop

closed 2024-11-19 by kilograham