raspberrypi/pico-sdk #1873: Problems building pico examples

seamusdemora, 2024-08-26:
Apparently I have bodged something... but I don't know what it is exactly.

I did an install of the pico sdk back in February of this year. I've not "pulled" any updates since then - until today. Since the `git pull(s)`, I cannot even build the `blink` example. Can someone tell me where I've "gone wrong"? 

```bash
20:29 : raspberrypi5 : ~/pico/pico-examples/build/blink
pi $ make -j2
PICO_SDK_PATH is /home/pi/pico/pico-sdk
Target board (PICO_BOARD) is 'pico'.
Using board configuration from /home/pi/pico/pico-sdk/src/boards/include/boards/pico.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
CMake Error at build/CMakeFiles/3.25.1/CMakeSystem.cmake:6 (include):
  include could not find requested file:

    /home/pi/pico/pico-sdk/cmake/preload/toolchains/pico_arm_gcc.cmake
Call Stack (most recent call first):
  /home/pi/pico/pico-extras/CMakeLists.txt:12 (project)

CMake Error at build/CMakeFiles/3.25.1/CMakeSystem.cmake:6 (include):
  include could not find requested file:

    /home/pi/pico/pico-sdk/cmake/preload/toolchains/pico_arm_gcc.cmake
Call Stack (most recent call first):
  CMakeLists.txt:7 (project)

Build type is Debug
Using regular optimized debug build (set PICO_DEOPTIMIZED_DEBUG=1 to de-optimize)
CMake Warning at /home/pi/pico/pico-sdk/tools/Findpicotool.cmake:28 (message):
  No installed picotool with version 2.0.0 found - building from source

  It is recommended to build and install picotool separately, or to set
  PICOTOOL_FETCH_FROM_GIT_PATH to a common directory for all your SDK
  projects
Call Stack (most recent call first):
  /home/pi/pico/pico-sdk/tools/CMakeLists.txt:138 (find_package)
  /home/pi/pico/pico-sdk/src/cmake/on_device.cmake:33 (pico_init_picotool)
  /home/pi/pico/pico-sdk/src/rp2040/boot_stage2/CMakeLists.txt:57 (pico_add_dis_output)
  /home/pi/pico/pico-sdk/src/rp2040/boot_stage2/CMakeLists.txt:101 (pico_define_boot_stage2)

Downloading Picotool
TinyUSB available at /home/pi/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
Compiling TinyUSB with CFG_TUSB_DEBUG=1
BTstack available at /home/pi/pico/pico-sdk/lib/btstack
cyw43-driver available at /home/pi/pico/pico-sdk/lib/cyw43-driver
lwIP available at /home/pi/pico/pico-sdk/lib/lwip
mbedtls available at /home/pi/pico/pico-sdk/lib/mbedtls
Skipping encrypted example which is unsupported on this platform
Skipping hello_dcp example which is unsupported on this platform
Skipping runtime_flash_permissions example which is unsupported on this platform
Skipping dvi_out_hstx_encoder example which is unsupported on this platform
Skipping spi_lcd example which is unsupported on this platform
Skipping multicore_doorbell example which is unsupported on this platform
Skipping hello_otp example which is unsupported on this platform
Skipping boot_info example which is unsupported on this platform
Skipping universal examples as PICO_RISCV_TOOLCHAIN_PATH and PICO_ARM_TOOLCHAIN_PATH are not defined
Skipping TinyUSB dual examples, as TinyUSB hw/mcu/raspberry_pi/Pico-PIO-USB submodule unavailable
Skipping SHA256 examples as pico_sha256 or pico_mbedtls unavailable
Skipping FreeRTOS examples as FREERTOS_KERNEL_PATH not defined
-- Configuring incomplete, errors occurred!
See also "/home/pi/pico/pico-examples/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/pico/pico-examples/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:1212: cmake_check_build_system] Error 1

20:30 : raspberrypi5 : ~/pico/pico-examples/build/blink
pi $
```

lurch, 2024-08-26:
As part of [the RP2350 release](https://www.raspberrypi.com/news/raspberry-pi-pico-2-our-new-5-microcontroller-board-on-sale-now/) there have been big updates to both [`pico-sdk`](https://github.com/raspberrypi/pico-sdk/releases/tag/2.0.0) and [`pico-examples`](https://github.com/raspberrypi/pico-examples/releases/tag/sdk-2.0.0).

My assumption is that you missed the bit where it says "**You should delete/recreate all build directories when upgrading from previous versions of the Raspberry Pi Pico SDK**" ?

seamusdemora, 2024-08-26:
@lurch : In fact I did miss that completely! I guess that was a post on the GitHub site?...  didn't see anything about it in my `git` output.

Thanks!

lurch, 2024-08-26:
> I guess that was a post on the GitHub site?

Yes, there's a link to it in my previous comment... :slightly_smiling_face:

seamusdemora, 2024-08-27:
Yes - I got it, and thanks again.

closed 2024-08-27 by seamusdemora