raspberrypi/pico-sdk #1876: cmake error

daharwalprashant, 2024-08-27:
[build] C:\windows\system32\cmd.exe /C "cd . && C:\PROGRA~2\GNUARM~1\102021~1.10\bin\AR014C~1.EXE -mcpu=cortex-m0plus -mthumb -Os -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fsingle-precision-constant -Wpedantic -Wall -std=gnu11 -mfloat-abi=soft -mlibarch=armv6s-m -march=armv6s-m -g  CMakeFiles/fain.dir/application/main.c.obj -o fain.exe -Wl,--out-implib,libfain.dll.a -Wl,--major-image-version,0,--minor-image-version,0  libfain_library.a  libtest_lib.a  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
[build] c:/progra~2/gnuarm~1/102021~1.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: unrecognized option '--major-image-version'
[build] c:/progra~2/gnuarm~1/102021~1.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: use the --help option for usage information
[build] collect2.exe: error: ld returned 1 exit status
[build] ninja: build stopped: subcommand failed.

the above error is been there while compiling. this string is not there in the workspace. what to do?

lurch, 2024-08-27:
Which instructions have you been following? You might want to start again from scratch, following the instructions in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf ? (There have been a **lot** of changes to pico-sdk recently https://github.com/raspberrypi/pico-sdk/releases/tag/2.0.0 )

kilograham, 2024-08-27:
Yeah, i have no idea what you are even trying to do.

kilograham, 2024-11-11:
closing as no response

closed 2024-11-11 by kilograham

Slion, 2025-03-17:
In my case I was missing the following:

```
# initialize the SDK based on PICO_SDK_PATH
# note: this must happen before project()
include(pico_sdk_import.cmake)
```