raspberrypi/pico-sdk #1877: 2.0.1 or hotfix release timeline

cramertj, 2024-08-27:
We're interested in picking up https://github.com/raspberrypi/pico-sdk/pull/1854 within the Pigweed repository, but we'd prefer not to shift off of the BCR (Bazel central registry) dependency and onto a git one. Would it be possible to get a hotfix release containing this change? I imagine the release timeline for 2.0.1 is sometime off.

If there's no objection, we could also consider pushing a BCR-only release (something like 2.0.0.bcr.1) containing this patch. This would result in BCR releases that do not correspond 1:1 with Pico SDK releases, but this approach could allow us to avoid a full SDK release cycle.

cc @kilograham @armandomontanez @tpudlik

tpudlik, 2024-08-27:
See also https://issues.pigweed.dev/issues/360950550#comment3 for more (maybe too much!) context.

armandomontanez, 2024-08-29:
It's worth noting that the Bazel build is currently broken so we may not want to be too hasty here. 😉

cramertj, 2024-09-25:
The Bazel build is now fixed, and I'd love to get a BCR release! :)
Let me know what way seems like the best option here.

kilograham, 2024-10-03:
Note this will likely be next week now; holler if this is a problem

closed 2024-10-03 by kilograham