raspberrypi/pico-sdk #1878: sm_config_set_in_pin_count() appears incorrect

arg08, 2024-08-28:
This function is newly added for RP2350, and the implementation appears inconsistent with the datasheet.

Datasheet says:
[quote]
Set the number of pins which are not masked to 0 when read by an
IN PINS, WAIT PIN or MOV x, PINS instruction.
For example, an IN_COUNT of 5 means that the 5 LSBs of the IN pin group are
visible (bits 4:0), but the remaining 27 MSBs are masked to 0. A count of 32 is
encoded with a field value of 0, so the default behaviour is to not perform any
masking.
[/quote]

ie. the number in the register is the number of pins you are actually using, except that zero represents  32.

However, the code puts `(in_count - 1)`.

https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2_common/hardware_pio/include/hardware/pio.h#L393

Looks like it should be `(in_count & 0x1f)`

peterharperuk, 2024-08-28:
@kilograham This does look wrong. Do you agree?

kilograham, 2024-08-28:
i do - have made a PR

kilograham, 2024-11-11:
merged into develop

PR #1880 (2024-08-29): fix bug in sm_config_set_in_pin_count
fixes #1878
closed 2024-11-11 by kilograham