raspberrypi/pico-sdk #1882: RP2350 partition table erased when reflashed

polhenarejos, 2024-08-29:
I am playing with partitions and I defined two (code & data) in my partition table. When I flash the board with the packed UF2 for the first time, everything works as expected. `picotool` reports 2 partitions (that's OK). But when I reflash the board again, the partition table is erased and the board is reboot to BOOTSEL mode. `picootol` displays it has no partitions. Then, I have to reflash again to work as expected.

It's a flip-flop behavior: the first time the partition table is int, the second time the partition table is erased, the third time appears again, and so on. Always I use the same UF2.

So, to summarize, I have to reflash the board twice to work normally, drag-and-drop .uf2 file twice.

Is this behavior normally expected?

CMake lines I used for partitions:
```
pico_set_uf2_family(${CMAKE_PROJECT_NAME} "rp2350-arm-s")
pico_embed_pt_in_binary(${CMAKE_PROJECT_NAME} "${CMAKE_CURRENT_LIST_DIR}/pt.json")
```

pt.json
```
{
  "version": [1, 0],
  "unpartitioned": {
    "families": ["absolute"],
    "permissions": {
      "secure": "rw",
      "nonsecure": "rw",
      "bootloader": "rw"
    }
  },
  "partitions": [
    {
      "name": "Test Firmware",
      "id": 0,
      "size": "1024K",
      "families": ["rp2350-arm-s", "rp2350-riscv"],
      "permissions": {
        "secure": "rw",
        "nonsecure": "rw",
        "bootloader": "rw"
      }
    },
    {
      "name": "Test Data",
      "id": 1,
      "size": "3072K",
      "families": ["data"],
      "permissions": {
        "secure": "rw",
        "nonsecure": "rw",
        "bootloader": "rw"
      },
      "link": ["owner", 0],
      "ignored_during_arm_boot": true,
      "ignored_during_riscv_boot": true
    }
  ]
}
```

When I say “reflash” I am referring to put the board in BOOTSEL mode and drag and drop the UF2 of my code (with PT embeded). I always use my uf2 all the time.

The thing is that if I download the same UF2 file multiple times, PT is erased alternate. One time picotool says no partitions there and the next time PT is correct and the next no, and next yes and so on.

As a sidenote, I tried the pico-example/flash/runtime_flash_permissions because is the example that uses a PT and it worked only once.
1. Nuke the board with pico-example/flash/nuke
2. Download runtime_flash_permissions.uf2
3. It works and executes normally (seen by UART0)
4. Put the board in BOOTSEL and drag-and-drop runtime_flash_permissions.uf2 again.
5. The board reboots in BOOTSEL immediately with no messages in UART0
6. Drag-and-drop runtime_flash_permissions.uf2 again and again but always reboots to BOOTSEL

I attach the .uf2 used in the tests
[runtime_flash_permissions.zip](https://github.com/user-attachments/files/16795976/runtime_flash_permissions.zip)
[flash_nuke.zip](https://github.com/user-attachments/files/16795979/flash_nuke.zip)
[pico_hsm.zip](https://github.com/user-attachments/files/16795980/pico_hsm.zip)

will-v-pi, 2024-08-29:
Yes, the reason for this behaviour is that you're embedding you are overwriting your partition table each time you upload, because it's embedded in your binary. Here are the steps that are happening

1. You start with no partition table, so your binary is loaded to the start of flash
2. Your partition table is in a block loop with your binary, so it is at address 0x10003C0C
4. You then open in bootsel mode and upload your binary again - this time there is a partition table, so it is placed according to that. It is placed in partition 0, so gets written to address 0x10002000, as that's the start address of your partition (that's the default if you don't specify a "start" in your JSON).
5. This now overwrites your partition table at 0x10003C0C, and invalidates the original block loop, so the device doesn't boot your binary and reboots straight to bootsel mode
6. There is no partition table again, so back to step 1

Having a partition table embedded in a binary is more of a bootloader thing - if you just want to have a partition table on your device then just run `picotool partition create pt.json pt.uf2` and upload that UF2 to your device, then upload your runtime_flash_permissions.uf2 and it'll be placed into partition 0 as expected and boot.

Alternatively, if you want to keep your PT embedded in your binary, but still be able to update your binary, then you can add `"start": 0` to the JSON for partition 0. This will still overwrite your partition table every time you upload, but because partition 0 now starts at 0x10000000 rather than 0x10002000, it won't cause these issues.

polhenarejos, 2024-08-29:
Very clever explanation. That was the issue and adding `"start":0` solves it.

closed 2024-08-29 by polhenarejos

kilograham, 2024-08-29:
Ah, I wasn't aware of the 0x2000 default, which is why i was confused (i had assumed it was 0x0 which would have worked fine). 

I guess this is reasonable, frankly we should suggest people use the absolute family when creating such a UF2