raspberrypi/pico-sdk #1887: add support for the riscv-none-elf triplet

ilg-ul, 2024-08-30:
The [xPack RISC-V Embedded GCC](https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/) toolchain uses the `riscv-none-elf` triplet.

If you agree, I would like to contribute a small pull request to add it to the `PICO_DEFAULT_GCC_TRIPLE` list in the CMake files.

kilograham, 2024-08-30:
sure; submit a PR

kilograham, 2024-08-30:
cc @Wren6991

ilg-ul, 2024-08-30:
> sure; submit a PR

done.

matsobdev, 2024-08-30:
I don't know if it will hold up, but since it has no `zcmp` (`_zcb_zcmp` fails) at least now, maybe adding an extra `pico_riscv_gcc_zcb.cmake`:
```
set(CMAKE_SYSTEM_PROCESSOR hazard3)

set(PICO_DEFAULT_GCC_TRIPLE riscv32-unknown-elf riscv32-corev-elf riscv-none-elf)

set(PICO_COMMON_LANG_FLAGS " -march=rv32ima_zicsr_zifencei_zba_zbb_zbs_zbkb_zca_zcb -mabi=ilp32")

include(${CMAKE_CURRENT_LIST_DIR}/util/pico_arm_gcc_common.cmake)
```
as well to differentiate impact on performance of user code, which is basically `pico_riscv_gcc_zcb_zcmp.cmake` without `zcmp`. Compiles, but if it works, I don't know.

ilg-ul, 2024-08-30:
The xPack RISC-V Embedded GCC toolchain is based on the upstream GCC, without changes. If there are extensions not yet upstreamed, I expect they'll be added at a certain point.

A friend edited locally these files and ran some benchmarks, thus I concluded that the xPack toolchain can be used to compile the SDK and some examples.

@matsobdev, I'm not sure what you suggest, but, if necessary, we can add a third file, no problem.

ilg-ul, 2024-09-01:
`zcmp` is not accepted by my 14.2.0-1 release because of the outdated binutils 2.42 (the error comes from the assembler).

I did a new build with binutils 2.43.1 and the `zcmp` extension was accepted:

```
ilg@wksi tmp % /Users/ilg/Work/xpack-dev-tools-build/riscv-none-elf-gcc-14.2.0-2/darwin-x64/application/bin/riscv-none-elf-gcc m.c -march=rv32imac_zicsr_zifencei_zba_zbb_zbs_zbkb_zca_zcb_zcmp -mabi=ilp32
ilg@wksi tmp % 
```

I'll make a new 14.2.0-2 release with the latest binutils.