raspberrypi/pico-sdk #1893: The sm_config_set_in_pins() set only one pin while it is named "pins".

suikan4github, 2024-09-01:
The sm_config_set_in_pins() function configures only one pin while it is named "pins".  Please kindly check. 

The definition of the sm_config_set_in_pins() is like this : 
https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2_common/hardware_pio/include/hardware/pio.h#L381-L383

This function receives the in_base parameter, and just passes it to the  sm_config_set_in_pin_base();

On the other hand, sm_config_set_out_pins() receives not only out_base but also out_count parameter. Then, calls     sm_config_set_out_pin_base()    sm_config_set_out_pin_count() internally, to configure the multiple pins. 

https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2_common/hardware_pio/include/hardware/pio.h#L304-L307

I think the implementation of the sm_config_set_out_pins() is right, and the implementation of the sm_config_set_in_pins() is wrong.

lurch, 2024-09-01:
> The sm_config_set_in_pins() function configures only one pin while it is named "pins". Please kindly check.

Perhaps [`sm_config_set_in_pins()`](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_sm_config_1gac418400e30520ea3961d8977c180a4f9) is just a "convenience function" which is named like that for commonality with [`sm_config_set_out_pins()`](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_sm_config_1gaf3004bbd996443d0c841664ebc92905c), [`sm_config_set_set_pins()`](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_sm_config_1gada1dff2c00b7d3a1cf722880c8373424), etc.? If you don't like the name, I guess you could just ignore it and _only_ use [`sm_config_set_in_pin_base()`](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_sm_config_1gaaf804a05a2ebaa127d0d216b3d4e3e80) instead? :wink: 

> the implementation of the sm_config_set_in_pins() is wrong.

I _think_ it's right - the ability to set the "count" of IN pins was a feature added in RP2350 (see section 11.1.1. in https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf), and so is configured via a separate [`sm_config_set_in_pin_count`](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_pio/include/hardware/pio.h#L385) function (which doesn't do anything on RP2040). And I assume that this wasn't added to `sm_config_set_in_pins()` in order to preserve backwards-compatibility?

suikan4github, 2024-09-01:
Thank you for the comment. 

> If you don't like the name, I guess you could just ignore it and only use [sm_config_set_in_pin_base()](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_sm_config_1gaaf804a05a2ebaa127d0d216b3d4e3e80) instead?

Yes. This is what am doing. 

 > The ability to set the "count" of IN pins was a feature added in RP2350

Ah, I could not notice it. So, I don't need to call sm_config_set_in_pin_count() for RP2040...

I understand current sm_config_set_in_pins() is left as is to provide backward compatibility. I agree not to change the parameters.

closed 2024-09-01 by suikan4github