raspberrypi/pico-sdk #1897: Why SWCLK and SWD register of pads_bank0 is not modeled in hardware structure?

oldoldman, 2024-09-02:
Hi guys,
From the rp2040 datasheet(page 300), there are 2 registers named SWCLK and SWD at the offset of PADS_BANK0_BASE+0x7C and PADS_BANK0_BASE+0x80, right after the register for GPIO30. But there is no place for SWCLK and SWD in [pads_bank0_hw_t](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2040/hardware_structs/include/hardware/structs/pads_bank0.h). In my understanding, either the sizeof pads_bank0_hw_t.io should be 32 or there should be 2 additional fields, for example,
```c++
typedef struct {
io_rw_32 voltage_select
io_rw_32 io[30];
io_rw_32 swclk;    // 
io_rw_32 swd;      //
} pads_bank0_hw_t;
```
How does this happen? Many thanks!

lurch, 2024-09-02:
Those `hardware_structs` are autogenerated, so pinging @matiasilva 

Interestingly, the "lower level" [rp2040/hardware_regs/include/hardware/regs/pads_bank0.h](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2040/hardware_regs/include/hardware/regs/pads_bank0.h) (which is also autogenerated) _does_ provide definitions for [SWCLK](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2040/hardware_regs/include/hardware/regs/pads_bank0.h#L2159) and [SWD](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2040/hardware_regs/include/hardware/regs/pads_bank0.h#L2230).

matiasilva, 2024-11-23:
Hi, these fields were not exposed in the [previous run of header generation](https://github.com/raspberrypi/pico-sdk/blob/68571ad33e42c8fcf39fadb0262a89556c1d0414/src/rp2040/hardware_structs/include/hardware/structs/padsbank0.h), so this looks intentional?

matiasilva, 2025-02-12:
My opinion is this can be safely closed/won't fix-ed as they were not exposed in the previous structs, presumably because nobody used them.

kilograham, 2025-02-12:
I don't believe it was deliberate.. cc @liamfraser, @Wren6991 .. we can punt this to 2.2.0 though since it's been this way forever