raspberrypi/pico-sdk #1899: PIO issues using new pio_claim_free_sm_and_add_program_for_gpio_range function

DangerousPrototypes, 2024-09-03:
Hello all,

Using Pico SDK 2.0 with RP2040 and RP2350 chips I'm experiencing unexpected behavior with PIO programs using the new ```pio_claim_free_sm_and_add_program_for_gpio_range``` function and ```pio_remove_program_and_unclaim_sm```.

[image]

Using the PIO I2C program in RP2350 with SDK 2.0 and new claim functions, this the result of a START and send 0xff. Garbled data, then a ton of what look like clock ticks on the data line.

[image]

Using the PIO I2C program in RP2040 with SDK 2.0 and new claim functions, this the result of a START and write 0xff. Garbled data, but not as crazy as RP2350.

[image]

This is using the PIO I2C program on RP2350 with SDK 2.0, except now I'm manually assigning PIO and SM. This is correct, and the same as RP2040 on SDK 1.5.

My suspicion is that the unclaim function is causing PIO instructions to be overwritten. I don't have simple code to reproduce this, but I have some observations:
- I'm switching PIO programs frequently, I don't think it happens if you just have one program at a time and don't replace it
- It seems to crash (panic) when the free state machine is between two claimed state machines. For example 0&2 are claimed, assigning something new to sm1 will crash the chip.

kilograham, 2024-09-03:
That is a fairly shoddy bug report. Did you try doing any debugging? Are you sure you are passing the right values to `pio_remove_program_and_unclaim_sm` ... maybe try printing out what SMs/offsets are claimed and freed and including that.

kilograham, 2024-09-03:
Also, you are using `_for_gpio_range`; what values are you passing for those?

kilograham, 2024-11-11:
closing as haven't heard back

closed 2024-11-11 by kilograham