raspberrypi/pico-sdk #1902: stdio_init_all() very slow in sdk 2.0

liubinbj, 2024-09-04:
in the old sdk 1.x , that function runs very fast, but in sdk 2.0 , it takes about 3s, it's funny. I guess it is a problem!

peterharperuk, 2024-09-04:
Is this to connect to usb or uart? Are you running one of the pico-examples?

liubinbj, 2024-09-04:
> Is this to connect to usb or uart? Are you running one of the pico-examples?

it connect to usb.  not examples, but one of my projects.

pico_enable_stdio_usb(skydogge 1)
pico_enable_stdio_uart(skydogge 0)

the stdio_init_all() is first line in the main(), but it takes 3 seconds, if use sdk 1.x , it's very fast.

liubinbj, 2024-09-04:
> PICO_STDIO_USB_CONNECT_WAIT_TIMEOUT_MS

don't know that and not set that value, and then I set  PICO_STDIO_USB_CONNECT_WAIT_TIMEOUT_MS=0 to test. still have 3s delay, with or without usb connection.

arturo182, 2024-09-04:
Probably best way to analyze would be to connect a debugger and stop execution during those 3s to see what the MCU doin.

peterharperuk, 2024-09-23:
Cannot reproduce. for me it takes about 0.5 ms. You'll have to attach an example that demonstrates the issue. The only reason it would take seconds is if it's waiting for usb to connect.

kilograham, 2024-11-11:
closing as not reproducable

closed 2024-11-11 by kilograham