raspberrypi/pico-sdk #1904: uf2 convert needs a -t elf appended to it's target argument

nitrologic, 2024-09-04:
to avoid this error in my pico2 project

````
C:\nitrologic\picotool\bin2\_deps\picotool\picotool uf2 convert --quiet C:/nitrologic/picotool/bin2/pico2tool pico2tool.uf2 --family rp2350-arm-s --abs-block""
ERROR: filename 'C:/nitrologic/picotool/bin2/pico2tool' does not have a recognized file type (extension)
ninja: build stopped: subcommand failed.
````

I needed to modify this line in the tools CMakeLists.txt

`                $<TARGET_FILE:${TARGET}> -t elf
`

https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/tools/CMakeLists.txt#L413

will-v-pi, 2024-09-05:
You likely have your imports in the wrong place in your CMakeLists.txt file - see #1870 

Namely the import of pico_sdk_init.cmake should be before the project statement

nitrologic, 2024-09-05:
@will-v-pi thank you, that fixed my issue!

closed 2024-09-05 by nitrologic