raspberrypi/pico-sdk #1905: where is the definition of cyw43_sdio_transfer_cmd53?

oldoldman, 2024-09-05:
Hi guys,
I can find the prototype of cyw43_sdio_transfer_cmd53 in ```lib\cyw43-driver\src\cyw43_sdio.h```, but no definition of this function in any c file except in ```lib\cyw43-driver\tests\sdio\mock_sdio.c``` which obviously is a fake implementation. Any suggestion? Many thank!

lurch, 2024-09-05:
This is probably a question for https://github.com/georgerobotics/cyw43-driver/issues ?

peterharperuk, 2024-09-05:
Not a pico-sdk issue. Closing.

Also check out the Micropython source where this is implemented for imx rt port

static inline int cyw43_sdio_transfer_cmd53(bool write, uint32_t block_size, uint32_t arg, size_t len, uint8_t *buf) {
    return sdio_transfer_cmd53(write, block_size, arg, len, buf);
}

oldoldman, 2024-09-05:
found it. ```cyw43_write_bytes``` interface is implemented by ```rp2_common\pico_cyw43_driver\cyw43_bus_pio_spi.c```.