raspberrypi/pico-sdk #1912: pico_get_unique_board_id_string prints an extra character at the start

LordOfDeadbush, 2024-09-06:
Hello,
I noticed in my code that for some reason `pico_get_unique_board_id_string()` is trimming the cstring incorrectly:

If I do this:
``` c++
#define BOARD_ID_LEN 17
char board_id[BOARD_ID_LEN];
pico_get_unique_board_id_string(board_id, BOARD_ID_LEN);
```

I get the following string:
E6625CA563857138

but if I do this: (if I change `BOARD_ID_LEN` to 16)
``` c++
#define BOARD_ID_LEN 16
char board_id[BOARD_ID_LEN];
pico_get_unique_board_id_string(board_id, BOARD_ID_LEN);
```

I get this string:
E6625CA56385713

weirdly enough, this fixes the problem:

``` c++
#define BOARD_ID_LEN 16
char board_id[BOARD_ID_LEN + 1];
pico_get_unique_board_id_string(board_id, BOARD_ID_LEN);
```

I looked in [https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_unique_id/unique_id.c](the source code) for this and found that there's an issue here:

```c++
    for (i = 0; (i < len - 1) && (i < PICO_UNIQUE_BOARD_ID_SIZE_BYTES * 2); i++) {
```

I think it should be `len` instead of  `len - 1` because we are using the less than operator, and not the LE operator. 

I can make a PR later today if needed!

Corn

LordOfDeadbush, 2024-09-06:
changed the title because I got mixed up with stuff :p

rhulme, 2024-09-07:
A normal string in C is null-terminated, so any buffer needs to be at least one byte bigger than the longest required string.
If `BOARD_ID_LEN` is 16, the buffer needs to be at least 17 bytes.  The second parameter to `pico_get_unique_board_id_string` is the length of the buffer you're passing in, not (directly) the length of the string you want.

The last line of the function writes the terminating NULL.  That's why the loop only copies to `len-1`.

lurch, 2024-09-07:
It seems @rhulme is correct, and indeed https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#group_pico_unique_id_1ga14fc6ab3ef89570b7dddff13e1b49634 actually says "For full serial, len >= 2 * PICO_UNIQUE_BOARD_ID_SIZE_BYTES + 1".