raspberrypi/pico-sdk #1915: pico_unique_id header needs some updates

lurch, 2024-09-07:
From reading the comments in [unique_id.h](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/pico_unique_id/include/pico/unique_id.h) (which also appear [here](https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#group_pico_unique_id) ) you might think that `pico_get_unique_board_id` **always** returns the ID of the flash chip. However looking at the [implementation](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/pico_unique_id/unique_id.c#L15) it looks like that on RP2350 it actually gives you the `SYS_INFO_CHIP_INFO` (which I think is described in section 5.4.8.17. of https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf ?).
And I guess this also means that you _won't_ get a unique_id of all-0xEE on PICO_NO_FLASH builds for RP2350? :wink:

peterharperuk, 2024-09-09:
Yes, it, it uses a value from OTP on rp2350

kilograham, 2024-09-28:
merged into devleop

PR #1942 (2024-09-28): Clarify that RP2350 gets its unique id from OTP
Fixes #1915
commit 66335f1d (2024-09-28): Clarify that RP2350 gets its unique id from OTP (#1942)

* Clarify that RP2350 gets its unique id from OTP

Fixes #1915

* Wrap rp2040_specific and rp2350_specific bits
closed 2024-09-28 by kilograham