raspberrypi/pico-sdk #1917: RP2040 build errors with Pico 2.0.0 SDK

iCMDdev, 2024-09-08:
I'm not entirely sure if this is an error with Pico SDK or [the CMakeLists.txt I'm using](https://github.com/apple/swift-embedded-examples/blob/22f2c981c524416fd4cf89ef84140a81cf2748b1/pico-blink-sdk/CMakeLists.txt), but when I try to build, I'm getting the following error:
```
pico-sdk/src/rp2_common/hardware_base/include/hardware/address_mapped.h:177:10: error: 'hardware/structs/accessctrl.h' file not found
175 | #if !PICO_RP2040
176 | // include this here to avoid the check in every other hardware/structs header that needs it
177 | #include "hardware/structs/accessctrl.h"
    |          `- error: 'hardware/structs/accessctrl.h' file not found
178 | #endif
179 | 
```
The hardware/structs/accessctrl.h` file is indirectly included by `pico/stdlib.h`.

Clearly, something's off: it seems that even though I'm compiling for RP2040 (the `PICO_BOARD` environment variable is set accordingly), the `#if !PICO_RP2040` fails. This does not occur in previous Pico SDK versions.

I figured this is somehow related to the `add_custom_command` and / or `add_dependencies` steps. In one case, removing the `add_dependencies` fixed this issue, but that's just a workaround, not a fix (if the dependency takes more time to compile, this will fail).

kilograham, 2024-09-08:
1. are you using a fresh build directory? - probably, but worth checking
2. where is this error coming from? - the SDK compile, or the swift bit
3. can you print the compile command if it is SDK

lurch, 2024-09-08:
> Edit: I assume [this is where](https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2040.cmake)`PICO_RP2040` is set (as a CMake variable, not a preprocessor macro though). Maybe this is related to a specific `target_compile_definitions`.

Just for the sake of completion, [here](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2040/pico_platform/CMakeLists.txt#L8) is where it gets set as a preprocessor macro.

iCMDdev, 2024-09-08:
> 1. are you using a fresh build directory? - probably, but worth checking

Great question. Yes.

> 2. where is this error coming from? - the SDK compile, or the swift bit

The SDK throws the error I mentioned, but the custom Swift parts added to the `CMakeLists.txt` seem to trigger it somehow... I'm not sure if it's a problem with the Swift compilation or withtin the SDK, that's the thing I'm trying to figure out. 

I've tried to replicate this in a C-only blink project using my own `add_custom_command`:

```cmake
cmake_minimum_required(VERSION 3.13)
include($ENV{PICO_SDK_PATH}/external/pico_sdk_import.cmake)

project(blink)

pico_sdk_init()

add_executable(blink)
add_custom_command(
    OUTPUT ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o
    COMMAND
        arm-none-eabi-gcc
        -c ${CMAKE_CURRENT_LIST_DIR}/blink_simple.c
        -o ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o
    DEPENDS
        ${CMAKE_CURRENT_LIST_DIR}/blink_simple.c
)
add_custom_target(swift-blinky-swiftcode DEPENDS ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o)

# pull in common dependencies
target_link_libraries(blink ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o)

add_dependencies(blink swift-blinky-swiftcode)

# create map/bin/hex file etc.
pico_add_extra_outputs(blink)
```

But this time, it works... So maybe this is related to Swift?

By the way, if I manually define `#define PICO_RP2040 1` inside the `hardware/structs/accessctrl.h` file, the build succeeds. The build also works fine for RP2350.

> 3. can you print the compile command if it is SDK

Sure, it's the usual cmake compile commands:

```terminal
$ cmake -G Ninja -B build .
$ cmake --build build
```

I also tried with make instead of Ninja, and the same error occurs.

iCMDdev, 2024-09-08:
> Just for the sake of completion, [here](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2040/pico_platform/CMakeLists.txt#L8) is where it gets set as a preprocessor macro.

Thanks!

kilograham, 2024-09-08:
> Sure, it's the usual cmake compile commands:
>
> $ cmake -G Ninja -B build .
> $ cmake --build build
>
> I also tried with make instead of Ninja, and the same error occurs.

Can you do:

```
$ cmake -G "Unix Makefiles" -B build .
$ cd build
$ make VERBOSE=1
```

and check the compile definitions

iCMDdev, 2024-09-08:
> > Sure, it's the usual cmake compile commands:
> > $ cmake -G Ninja -B build .
> > $ cmake --build build
> > I also tried with make instead of Ninja, and the same error occurs.
> 
> Can you do:
> 
> ```
> $ cmake -G "Unix Makefiles" -B build .
> $ cd build
> $ make VERBOSE=1
> ```
> 
> and check the compile definitions

 Yes, sure. It's properly added to the `arm-none-eabi-gcc`:

```console
[  1%] Building ASM object pico-sdk/src/rp2040/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
cd /Users/user/pico/examples/swift-embedded-examples/pico-blink-sdk/build/pico-sdk/src/rp2040/boot_stage2 && /opt/homebrew/bin/arm-none-eabi-gcc -DPICO_32BIT=1 -DPICO_BOARD=\"pico\" -DPICO_BUILD=1 -DPICO_NO_HARDWARE=0 -DPICO_ON_DEVICE=1 -DPICO_RP2040=1 -I/Users/user/pico/pico-sdk/src/rp2040/boot_stage2/asminclude -isystem /Users/user/pico/pico-sdk/src/rp2040/hardware_regs/include -isystem /Users/user/pico/pico-sdk/src/rp2_common/hardware_base/include -isystem /Users/user/pico/pico-sdk/src/common/pico_base_headers/include -isystem /Users/user/pico/examples/swift-embedded-examples/pico-blink-sdk/build/generated/pico_base -isystem /Users/user/pico/pico-sdk/src/boards/include -isystem /Users/user/pico/pico-sdk/src/rp2040/pico_platform/include -isystem /Users/user/pico/pico-sdk/src/rp2_common/pico_platform_compiler/include -isystem /Users/user/pico/pico-sdk/src/rp2_common/pico_platform_panic/include -isystem /Users/user/pico/pico-sdk/src/rp2_common/pico_platform_sections/include -isystem /Users/user/pico/pico-sdk/src/rp2040/boot_stage2/include -mcpu=cortex-m0plus -mthumb -g -O3 -DNDEBUG -o CMakeFiles/bs2_default.dir/compile_time_choice.S.obj   -c /Users/user/pico/pico-sdk/src/rp2040/boot_stage2/compile_time_choice.S
```

But, if I look further into this, it crashes at `Generating _swiftcode.o`:
```make
/usr/bin/swiftc -target armv6m-none-none-eabi -Xcc -mfloat-abi=soft -Xcc -fshort-enums -Xfrontend -function-sections -enable-experimental-feature Embedded -wmo -parse-as-library $( echo '/Users/user/pico/pico-sdk/src/common/pico_stdlib_headers/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_gpio/include;/Users/user/pico/pico-sdk/src/common/pico_base_headers/include;/Users/user/pico/examples/swift-embedded-examples/pico-blink-sdk/build/generated/pico_base;/Users/user/pico/pico-sdk/src/boards/include;/Users/user/pico/pico-sdk/src/rp2040/pico_platform/include;/Users/user/pico/pico-sdk/src/rp2040/hardware_regs/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_base/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_platform_compiler/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_platform_panic/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_platform_sections/include;/Users/user/pico/pico-sdk/src/rp2040/hardware_structs/include;/Users/user/pico/pico-sdk/src/common/hardware_claim/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_sync/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_sync_spin_lock/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_irq/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_uart/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_resets/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_clocks/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_pll/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_vreg/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_watchdog/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_ticks/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_xosc/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_divider/include;/Users/user/pico/pico-sdk/src/common/pico_time/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_timer/include;/Users/user/pico/pico-sdk/src/common/pico_sync/include;/Users/user/pico/pico-sdk/src/common/pico_util/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_time_adapter/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_runtime/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_runtime_init/include;/Users/user/pico/pico-sdk/src/common/pico_bit_ops_headers/include;/Users/user/pico/pico-sdk/src/common/pico_divider_headers/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_double/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_float/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_malloc/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_atomic/include;/Users/user/pico/pico-sdk/src/common/pico_binary_info/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_printf/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_stdio/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_stdio_uart/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_bootrom/include;/Users/user/pico/pico-sdk/src/common/boot_picoboot_headers/include;/Users/user/pico/pico-sdk/src/rp2_common/hardware_boot_lock/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_int64_ops/include;/Users/user/pico/pico-sdk/src/rp2_common/pico_mem_ops/include;/Users/user/pico/pico-sdk/src/rp2040/boot_stage2/include;/Users/user/pico/pico-sdk/src/common/boot_picobin_headers/include' | tr ';' '\n' | sed -e 's/\(.*\)/-Xcc -I\1/g' ) $( echo '/Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/lib/gcc/arm-none-eabi/13.3.1/include /Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/lib/gcc/arm-none-eabi/13.3.1/include-fixed /Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/arm-none-eabi/include' | tr ' ' '\n' | sed -e 's/\(.*\)/-Xcc -I\1/g' ) -i [... cut ...]
make[1]: *** [CMakeFiles/swift-blinky-swiftcode.dir/all] Error 2
make: *** [all] Error 2
```

I think I figured out a possible reason for all of this: this build step is the custom Swift. The "main" file of this project is a Swift file: `Main.swift` uses `BridgingHeader.h` to include `pico/stdlib.h`, but there's no Pico SDK compile definitions passed to it. And this is obviously because the Pico SDK doesn't handle this file on its own, it doesn't manage its compilation. So I'd have to provide these compile definitions on my own. Right?

kilograham, 2024-09-09:
Yes, this is rather what i thought might be happening, but you had said it wasn't coming from the swift bit

The swift-embedded should be using the compile definitions for the headers the same as for the regular build, or you may get inconsistencies - this was the case before, but here you now have a compile failure rather than potential silent weirdness

You can either set explicitly, do something hacky like https://github.com/raspberrypi/pico-playground/blob/9930bb057b7272176fbcd924b1d5856dfbc07503/standalone/static_sdk/CMakeLists.txt#L57 (to recursively gather the non-generator-expression compile definitions off your binary) or see if you can correctly get CMake to pass the compile definitions to your swift header import

iCMDdev, 2024-09-09:
All right, thanks a lot for your help & suggestions! I closed this as it isn't an SDK issue.

closed 2024-09-09 by iCMDdev

iCMDdev, 2024-09-09:
I tried the pico-playground example and it works flawlessly! 👍
Thanks alot.