raspberrypi/pico-sdk #1922: RP2350 RISC-V compilation fails due to inline asm machine constraint - platform.h

iCMDdev, 2024-09-11:
When trying to compile a program using LLVM's clang for the RISC-V architecture of RP2350, an error is thrown at [`/src/rp2350/pico_platform/include/pico/platform.h:249:31`](https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2350/pico_platform/include/pico/platform.h#L249):
```h
error: invalid output constraint '+l' in asm
247 | __force_inline static int32_t __mul_instruction(int32_t a, int32_t b) {
248 | #ifdef __riscv
249 |     __asm ("mul %0, %0, %1" : "+l" (a) : "l" (b) : );
    |                               `- error: invalid output constraint '+l' in asm
250 | #else
251 |     pico_default_asm ("muls %0, %1"     : "+l" (a) : "l" (b) : "cc");
```

Looking at [GCC's machine constraints for RISC-V](https://gcc.gnu.org/onlinedocs/gcc/Machine-Constraints.html), there isn't any `l` constraint specific for RISC-V, and [there's no general `l` constraint](https://gcc.gnu.org/onlinedocs/gcc/Simple-Constraints.html). This also goes for LLVM.

This leads me to believe this was a mistake, since `l` is valid on ARM and was previously used for the ARM compilation mode. I'm not sure why / how this compilation succeeds on GCC, even though `l` is an invalid constraint for RISC-V. 

One option is to replace the `l` constraint for RISC-V with `r` like this:
```h
247 | __force_inline static int32_t __mul_instruction(int32_t a, int32_t b) {
248 | #ifdef __riscv
249 |     __asm ("mul %0, %0, %1" : "+r" (a) : "r" (b) : );
250 | #else
251 |     pico_default_asm ("muls %0, %1"     : "+l" (a) : "l" (b) : "cc");
```

iCMDdev, 2024-09-11:
Makes sense, I might not have used it in my testing. Thanks for merging!

closed 2024-09-11 by iCMDdev

kilograham, 2024-09-11:
Yeah; i mean i think it is the same on Clang - we do check Clang compilation; i imagine it may be related to swift bridge headers - in any case the fix is in