raspberrypi/pico-sdk #1929: Compilation issue after upgrading to 2.0.0 from 1.5.1

leondisctech, 2024-09-13:
I have a project where I have the Pico SDK in a git submodule. I recently tried upgrading the SDK from version 1.5.1 to version 2.0.0, and now I have compilation issues where runtime_init.h isn't found. The following is from running cmake and make in a clean build directory. (I deleted the build directory as instructed in the SDK 2.0.0 release notes.) The error is about 7 lines from the bottom. (`/Users/foo/bar/pico-sdk/src/common/pico_sync/mutex.c:9:10: fatal error: pico/runtime_init.h: No such file or directory`)

```
foo@computer build % cmake ..
PICO_SDK_PATH is /Users/foo/bar/pico-sdk
Defaulting platform (PICO_PLATFORM) to 'rp2040' since not specified.
Defaulting target board (PICO_BOARD) to 'pico' since not specified.
Using board configuration from /Users/foo/bar/pico-sdk/src/boards/include/boards/pico.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
-- Defaulting build type to 'Release' since not specified.
Defaulting compiler (PICO_COMPILER) to 'pico_arm_cortex_m0plus_gcc' since not specified.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m0plus_gcc'
Defaulting PICO_GCC_TRIPLE to 'arm-none-eabi'
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/local/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/local/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /opt/local/bin/arm-none-eabi-gcc
Build type is Release
CMake Warning at pico-sdk/tools/Findpicotool.cmake:28 (message):
  No installed picotool with version 2.0.0 found - building from source

  It is recommended to build and install picotool separately, or to set
  PICOTOOL_FETCH_FROM_GIT_PATH to a common directory for all your SDK
  projects
Call Stack (most recent call first):
  pico-sdk/tools/CMakeLists.txt:138 (find_package)
  pico-sdk/src/cmake/on_device.cmake:33 (pico_init_picotool)
  pico-sdk/src/rp2040/boot_stage2/CMakeLists.txt:57 (pico_add_dis_output)
  pico-sdk/src/rp2040/boot_stage2/CMakeLists.txt:101 (pico_define_boot_stage2)

Downloading Picotool
-- Found Python3: /opt/local/bin/python3.12 (found version "3.12.5") found components: Interpreter
TinyUSB available at /Users/foo/bar/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /Users/foo/bar/pico-sdk/lib/btstack
cyw43-driver available at /Users/foo/bar/pico-sdk/lib/cyw43-driver
lwIP available at /Users/foo/bar/pico-sdk/lib/lwip
mbedtls available at /Users/foo/bar/pico-sdk/lib/mbedtls
-- Configuring done (2.0s)
-- Generating done (0.2s)
-- Build files have been written to: /Users/foo/bar/build
foo@computer build % make
[  0%] Creating directories for 'pioasmBuild'
[  0%] No download step for 'pioasmBuild'
[  1%] No update step for 'pioasmBuild'
[  1%] No patch step for 'pioasmBuild'
[  1%] Performing configure step for 'pioasmBuild'
Not searching for unused variables given on the command line.
loading initial cache file /Users/foo/bar/build/pico-sdk/src/rp2_common/pico_cyw43_driver/pioasm/tmp/pioasmBuild-cache-Release.cmake
-- The CXX compiler identification is AppleClang 15.0.0.15000309
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done (0.3s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/foo/bar/build/pioasm
[  1%] Performing build step for 'pioasmBuild'
[100%] Built target pioasm
[  1%] Performing install step for 'pioasmBuild'
[100%] Built target pioasm
Install the project...
-- Install configuration: "Release"
[  1%] Completed 'pioasmBuild'
[  1%] Built target pioasmBuild
[  1%] Generating rp2040_sdio.pio.h
[  1%] Built target no-OS-FatFS-SD-SDIO-SPI-RPi-Pico_rp2040_sdio_pio_h
[  1%] Built target picotoolForceReconfigure
[  1%] Creating directories for 'picotoolBuild'
[  1%] No download step for 'picotoolBuild'
[  1%] No update step for 'picotoolBuild'
[  1%] No patch step for 'picotoolBuild'
[  2%] Performing configure step for 'picotoolBuild'
Not searching for unused variables given on the command line.
-- The C compiler identification is AppleClang 15.0.0.15000309
-- The CXX compiler identification is AppleClang 15.0.0.15000309
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
.//Users/foo/bar/build/_deps/picotool/
-- Using the single-header code from /Users/foo/bar/build/_deps/picotool-src/lib/nlohmann_json/single_include/
CMake Deprecation Warning at /Users/foo/bar/pico-sdk/lib/mbedtls/CMakeLists.txt:23 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found Python3: /Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/bin/python3.9 (found version "3.9.6") found components: Interpreter
libUSB is not found - no USB support will be built
-- Configuring done (0.6s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/foo/bar/build/_deps/picotool-build
[  2%] Performing build step for 'picotoolBuild'
[ 67%] Built target mbedcrypto
[ 75%] Built target mbedx509
[ 85%] Built target mbedtls
[ 87%] Built target errors
[ 88%] Built target elf
[ 91%] Built target bintool
[ 93%] Built target elf2uf2
ld: warning: ignoring duplicate libraries: 'errors/liberrors.a'
[100%] Built target picotool
[  2%] Performing install step for 'picotoolBuild'
[ 67%] Built target mbedcrypto
[ 75%] Built target mbedx509
[ 85%] Built target mbedtls
[ 87%] Built target errors
[ 88%] Built target elf
[ 91%] Built target bintool
[ 93%] Built target elf2uf2
[100%] Built target picotool
Install the project...
-- Install configuration: "Release"
[  2%] Performing test step for 'picotoolBuild'
picotool v2.0.0 (Darwin, AppleClang-15.0.0.15000309, Release)
[  2%] Completed 'picotoolBuild'
[  2%] Built target picotoolBuild
[  2%] Building ASM object pico-sdk/src/rp2040/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  2%] Linking ASM executable bs2_default.elf
[  2%] Built target bs2_default
[  3%] Generating bs2_default.bin
[  3%] Generating bs2_default_padded_checksummed.S
[  3%] Built target bs2_default_padded_checksummed_asm
[  3%] Building C object helpers/CMakeFiles/mutex.dir/__/pico-sdk/src/common/pico_sync/sem.c.obj
[  3%] Building C object helpers/CMakeFiles/mutex.dir/__/pico-sdk/src/common/pico_sync/lock_core.c.obj
[  3%] Building C object helpers/CMakeFiles/mutex.dir/__/pico-sdk/src/common/pico_sync/mutex.c.obj
/Users/foo/bar/pico-sdk/src/common/pico_sync/mutex.c:9:10: fatal error: pico/runtime_init.h: No such file or directory
    9 | #include "pico/runtime_init.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [helpers/CMakeFiles/mutex.dir/__/pico-sdk/src/common/pico_sync/mutex.c.obj] Error 1
make[1]: *** [helpers/CMakeFiles/mutex.dir/all] Error 2
make: *** [all] Error 2
```

Running cmake/build thru visual studio has some different flags but produces similar output except that it gives the error both in mutex.c and time.c.

I'm guessing this is just me goosing up git submodules and/or cmake somehow, but I'd appreciate any guidance.

kilograham, 2024-09-16:
The include directories should be correct by default, so there is something else messed up with your build environment(s)

can you post `CMakeLists.txt`

sakhnik, 2024-09-16:
Sure.
[k40-rpi-pico.zip](https://github.com/user-attachments/files/17017325/k40-rpi-pico.zip)
I'm trying to build example from the the article: https://embeddedcomputing.com/technology/open-source/linux-freertos-related/using-freertos-with-the-raspberry-pi-pico, but with FetchContent.

kilograham, 2024-09-16:
Hmm, can you try with https://github.com/raspberrypi/FreeRTOS-Kernel instead (the upstream does not have latest fixes or SDK 2.0.0 related changes, but i wasn't aware it was broken with SDK 2.0.0)

sakhnik, 2024-09-16:
Unfortunately, still the same error.
Even if I manually specify the include path `include_directories($ENV{PICO_SDK_PATH}/src/host/pico_runtime/include)`, there are still compilation errors.

kilograham, 2024-09-16:
> Even if I manually specify the include path include_directories($ENV{PICO_SDK_PATH}/src/host/pico_runtime/include), 

yeah, i mean that shouldn't be the fix

kilograham, 2024-09-16:
Oh, well, yeah.. not sure where people came up with that method of pulling in FreeRTOS_Kernel

You should use FreeRTOS_Kernel_Import.cmake (like pico-sdk-import)

leondisctech, 2024-09-16:
Here's my `CmakeLists.txt` (library and directory names changed for privacy):

```
cmake_minimum_required(VERSION 3.20...3.27)

# initialize pico-sdk from submodule
# note: this must happen before project()
include(pico-sdk/pico_sdk_init.cmake)

project(my_project)
set(CMAKE_CXX_STANDARD 23)

# initialize the Raspberry Pi Pico SDK
pico_sdk_init()

# gcc spits about a bunch of inactionable ABI warnings about strings. no-psabi supresses them.
add_compile_options(-Wno-psabi)

add_subdirectory(dir1)
add_subdirectory(dir2)
add_subdirectory(dir3)
add_subdirectory(dir4)

add_executable(main
    main.cpp
    other.cpp
)

# Add pico_stdlib library which aggregates commonly used features
target_link_libraries(main
    lib1
    lib2
    lib3
    lib4
    lib5
    lib6
    lib7
    lib8
    lib9
    lib10
    lib11
    pico_stdlib
    lib12
)

# enable usb output, enable uart output
pico_enable_stdio_usb(main 1)
pico_enable_stdio_uart(main 1)

# create map/bin/hex/uf2 file in addition to ELF.
pico_add_extra_outputs(main)
```

leondisctech, 2024-09-17:
And just to reiterate: the above CMakeLists works with SDK 1.5.1 but not with 2.0.0. If I upgrade to 2.0.0, it doesn't work. If I downgrade back to 1.5.1, it works. I can't share the whole codebase, but I can try to cook up something that I can share (closer to a minimal working example).

Poofjunior, 2024-09-17:
Same issue with this library we wrote here actually. Compilation works just fine under 1.5.1 but fails with the same error above in 2.0.0.

```bash
[  1%] Building C object build/CMakeFiles/usb_desc.dir/src/usb_descriptors.c.obj
[  1%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/dcd_rp2040.c.obj
[  1%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c.obj
[  1%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/device/usbd.c.obj
[  2%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/device/usbd_control.c.obj
[  2%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/class/audio/audio_device.c.obj
[  2%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/class/cdc/cdc_device.c.obj
[  3%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/class/dfu/dfu_device.c.obj
[  3%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/class/dfu/dfu_rt_device.c.obj
[  3%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/class/hid/hid_device.c.obj
[  3%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/class/midi/midi_device.c.obj
[  4%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/class/msc/msc_device.c.obj
[  4%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/class/net/ecm_rndis_device.c.obj
[  4%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/class/net/ncm_device.c.obj
[  4%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/class/usbtmc/usbtmc_device.c.obj
[  5%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/class/vendor/vendor_device.c.obj
[  5%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/class/video/video_device.c.obj
[  5%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/tusb.c.obj
[  6%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/lib/tinyusb/src/common/tusb_fifo.c.obj
[  6%] Building C object build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/src/rp2_common/hardware_irq/irq.c.obj
/home/poofjunior/external_projects/pico-sdk/src/rp2_common/hardware_irq/irq.c:8:10: fatal error: pico/runtime_init.h: No such file or directory
    8 | #include "pico/runtime_init.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [build/CMakeFiles/usb_desc.dir/build.make:342: build/CMakeFiles/usb_desc.dir/home/poofjunior/external_projects/pico-sdk/src/rp2_common/hardware_irq/irq.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:2013: build/CMakeFiles/usb_desc.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

* [public library repo](https://github.com/AllenNeuralDynamics/harp.core.rp2040)
  * library [CMakeLists.txt](https://github.com/AllenNeuralDynamics/harp.core.rp2040/blob/main/firmware/CMakeLists.txt)
* [example](https://github.com/AllenNeuralDynamics/harp.core.rp2040/tree/main/examples/harp_c_app_example) using the library
  * example [CMakeLists.txt](https://github.com/AllenNeuralDynamics/harp.core.rp2040/blob/main/examples/harp_c_app_example/CMakeLists.txt)

It's worth mentioning that we are invoking TinyUSB calls directly and needed to pull out usb_descriptors.c in the process.

sakhnik, 2024-09-18:
I'm now able to build with both pico SDK 1.5.1 and 2.0.0. I use FetchContent_Populate to get pico SDK and FreeRTOS SDK. I include their import cmake files before declaring my project and everything works out.
[k40-rpi-pico-1.zip](https://github.com/user-attachments/files/17040326/k40-rpi-pico-1.zip)

leondisctech, 2024-09-18:
@sakhnik Glad you got it working! Do you have any insight into why those changes help? (I'm trying to figure out if there are any lessons for how to do this without FetchContent.)

sakhnik, 2024-09-18:
@leondisctech, nothing specific. I just followed the README from freertos RP2040 port. They suggest including their cmake module before project(), and that's it. I didn't try without FetchContent.

Poofjunior, 2024-09-20:
Ok, it looks like tweaking the parent library's CMakeLists.txt fixed it.
```cmake
# old
target_link_libraries(usb_desc tinyusb_device pico_unique_id)
# new
target_link_libraries(usb_desc tinyusb_device pico_unique_id pico_stdlib)
```

It looks like we did something eyebrow-raising by ripping out usb_desc and trying to compile its dependencies alone, but adding `pico_stdlib` as a dependency appears to have fixed it.

leondisctech, 2024-09-20:
@Poofjunior Thanks for the tip! Adding pico_stdlib did the trick.

@kilograham Any idea why [this file](https://github.com/Discordant-Technologies-LLC/sdk_problem/blob/master/helpers/mutex.h) now needs both `pico_sync` and `pico_stdlib` in CMakeLists under pico-sdk 2.0.0 while it only needed `pico_sync` under pico-sdk 1.5.1?

kilograham, 2024-09-20:
It needs `pico_runtime_init`... it is a bug that it doesn't depend on it directly

aabizri, 2024-10-07:
I hit the same issue, I've made PR #1956 that fixes these compile errors in my testing

kilograham, 2024-11-09:
latest PR merged into develop

PR #2031 (2024-11-09): Fix some header dependencies
Fixes #1929
commit 809a423f (2024-11-09): Fix some header dependencies (#2031)

Fixes #1929
closed 2024-11-09 by kilograham