raspberrypi/pico-sdk #1930: How can i correctly install picotool on Linux

erfanium, 2024-09-14:
I have installed picotool binary and it is available within my $PATH:
```sh
$ picotool
PICOTOOL:
    Tool for interacting with RP2040/RP2350 device(s) in BOOTSEL mode, or with an RP2040/RP2350 binary......

```

But every time i want build a pico project, I get this warning message:

```sh
CMake Warning at /home/erfanium/Documents/pico-sdk/tools/Findpicotool.cmake:28 (message):
  No installed picotool with version 2.0.0 found - building from source

  It is recommended to build and install picotool separately, or to set
  PICOTOOL_FETCH_FROM_GIT_PATH to a common directory for all your SDK
  projects
Call Stack (most recent call first):
  /home/erfanium/Documents/pico-sdk/tools/CMakeLists.txt:138 (find_package)
  /home/erfanium/Documents/pico-sdk/src/cmake/on_device.cmake:33 (pico_init_picotool)
  /home/erfanium/Documents/pico-sdk/src/rp2040/boot_stage2/CMakeLists.txt:57 (pico_add_dis_output)
  /home/erfanium/Documents/pico-sdk/src/rp2040/boot_stage2/CMakeLists.txt:101 (pico_define_boot_stage2)
```

It says `It is recommended to build and install picotool separately`, but there is no documentation about it, How should i really install it?

will-v-pi, 2024-09-14:
See the [picotool README](https://github.com/raspberrypi/picotool?tab=readme-ov-file#usage-by-the-raspberry-pi-pico-sdk) for instructions - there have been some improvements to the inclusion on some platforms since release, so if it doesn’t work for you on the master branch then try using the develop branch of picotool.

If it doesn’t work with the develop branch, then can you give more details of what platform you are on?

erfanium, 2024-09-14:
I reinstalled the picotool and now it's working.

seams the binary location should be `/usr/local/bin/picotool`, not `/usr/bin/picotool` is my case

lurch, 2024-09-17:
I guess this issue can be closed then @erfanium ?

erfanium, 2024-09-19:
yes

closed 2024-09-19 by erfanium