raspberrypi/pico-sdk #1932: USB serial locks up under load

ali1234, 2024-09-15:
To reproduce, build and flash this code to a Pico (not Pico W because the LED won't blink):

```
#include <stdio.h>
#include <pico/stdlib.h>

#define LED_PIN 25

int main() {
    stdio_init_all();
    gpio_init(LED_PIN);
    gpio_set_dir(LED_PIN, GPIO_OUT);
    while(true) {
        gpio_put(LED_PIN, 0);
        getchar();
        gpio_put(LED_PIN, 1);
    }
}
```

Now on PC run this, redirecting to whatever serial port the Pico is on:

```
cat /dev/zero | pv > /dev/ttyACM0
```

Expected result: the LED blinks whenever a character is received, for as long as you leave `cat` running.

Actual result: the LED stops blinking after about 1 to 10 seconds. `pv` shows that data is no longer moving - the port is stalled because it thinks the buffer is full. But the pico is deadlocked inside `getchar()` presumably because it thinks the buffer is empty. If you run `cat` again, it will blink again, for a while, and then lock up again.

This seems easier to reproduce in real code on a RP2350, but example code fails on RP2040 as well.

ali1234, 2024-09-16:
While attempting to debug this I accidentally left the Pico halted in the debugger which triggers the "XHCI died" linux bug, disabling all USB ports on the host computer and forcing a reboot. After the reboot I could no longer reproduce the serial locking up. I didn't even reflash the Pico after rebooting so it is definitely the same firmware. So this seems like a Linux serial port bug.

peterharperuk, 2024-09-20:
I tried your example and couldn't reproduce it either.

mordae, 2024-11-02:
This hangs for me randomly after a day or so:

```
#include <pico/stdlib.h>
#include <pico/stdio_usb.h>

#if !defined(PICO_DEFAULT_LED_PIN)
#error blink example requires a board with a regular LED
#endif

#define LED_PIN PICO_DEFAULT_LED_PIN

int main()
{
	stdio_usb_init();

	gpio_init(LED_PIN);
	gpio_set_dir(LED_PIN, GPIO_OUT);
	gpio_put(LED_PIN, 1);

	for (int i = 0; i < 20; i++) {
		if (stdio_usb_connected())
			break;

		gpio_put(LED_PIN, i & 1);
		sleep_ms(100);
	}

	gpio_put(LED_PIN, 0);

	while (true) {
		int c = getchar_timeout_us(100);

		if ('W' == c)
			gpio_put(LED_PIN, 1);
		else if ('w' == c)
			gpio_put(LED_PIN, 0);

		sleep_ms(50);
	}
}
```

kilograham, 2024-11-02:
try with the `develop` branch; there was a timer bug fixed there which might be the cause

kilograham, 2026-09-30:
I think this was the timer bug fixed since, so closing.

closed 2026-09-30 by kilograham